platformio/platform-espressif32 #915: TTGO T-Display-S3 support sanity check

millardjn, 2022-09-16:
Hello I have been learning to use platformio with a TTGO TDisplayS3 https://github.com/Xinyuan-LilyGO/T-Display-S3

I have managed to get the board working without any checksum errors or issues with my projects so far with the following definition:
```json
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
	  "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DARDUINO_USB_CDC_ON_BOOT=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
	  "-DBOARD_HAS_PSRAM"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "TDisplay Espressif ESP32-S3 (16 MB QD, 8MB PSRAM)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 8716288,
    "maximum_size": 16777216,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://github.com/Xinyuan-LilyGO/T-Display-S3",
  "vendor": "Espressif"
}
```
I'm not 100% confident in the above as I have cobbled together this config by copying from similar boards. If anyone can give me some steps to check that the board works correctly with the above I'll be more than happy to test it out so this can be included in the list of supported boards.

Jason2866, 2022-09-16:
This is not needed:
```
-DARDUINO_RUNNING_CORE=1
-DARDUINO_EVENT_RUNNING_CORE=1
use_1200bps_touch": true,
```
and change entry to
```
maximum_ram_size": 327680,
```
and you can add in the according sections
```
-DARDUINO_USB_MODE=1

"connectivity": [
"wifi",
"bluetooth",
"ethernet"
],
```

millardjn, 2022-09-17:
Thanks, I've made those changes.

I ran into issues when removing `"use_1200bps_touch": true` without also removing `"wait_for_upload_port": true`. Seems like either both or neither were required so I've removed both.

Regarding PSRAM some discussions https://github.com/platformio/platform-espressif32/issues/596 seem to indicate that the PSRAM amount should be added to `maximum_ram_size`, but other board configurations such as `Adafruit Feather ESP32-S3 2MB PSRAM` don't and just have 327680 as you've said. Is it just the  "memory_type": "qio_opi" required to set up PSRAM?

The `-DARDUINO_RUNNING_CORE=1` `-DARDUINO_EVENT_RUNNING_CORE=1` lines were taken from the `Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)` board file. Removing them hasn't caused any problems that i can tell. Whats the difference between the two boards?

with `-DARDUINO_USB_MODE=1` I am no longer losing the first couple of serial lines which is nice, but it has revealed:
```
Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
```
Which isn't stopping function but seems like it isn't good. What usually causes this?

<details>
<summary>updated json</summary>

```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DARDUINO_USB_CDC_ON_BOOT=1",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "T-Display Espressif ESP32-S3 (16 MB QD, 8MB PSRAM)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://github.com/Xinyuan-LilyGO/T-Display-S3",
  "vendor": "Espressif"
}
```

</details>

Jason2866, 2022-09-17:
The entry `maximum_ram_size` has to be there, but is unused. Every value will work.
This is normal
```
Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
```
without the entry, you just did not see it. Since Arduino uses precompiled bootloaders where no checksum is there (fffff...) It is compared to the calculated when booting. The can not match. So the message. This function do have only the S3 and the C3.

commit b94ae136 (2023-02-27): Update Arduino core to v2.0.7

- New boards (resolves #915, resolves #1009)
- Updated IDF to v4.4.4 for mixed projects
closed 2023-02-27 by valeros