platformio/platform-espressif32 #916: Please update the esp-idf framework to include adc_continious.c

zapta, 2022-09-16:
Currently platformio provides the version below which doesn't include support the support library for continuous adc reading.

framework-espidf @ 3.40401.0 (4.4.1)

https://github.com/espressif/esp-idf/commits/master/components/esp_adc/adc_continuous.c

Jason2866, 2022-09-16:
Branch develop supports IDF 4.4.2

zapta, 2022-09-16:
Adding this the my platformio.ini

platform_packages = framework-espidf @ ~3.40402.0

Resulted in set of versions and grepping my ~/.platformio/packages to adc_continious yielded no results.

```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.1.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40402.0 (4.4.2)   <<===
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
```

Jason2866, 2022-10-02:
Ahh, missed it is idf branch master. There is currently no support for IDF 5.0
There is no espressif release version. Platformio does not support unreleased versions.

hselomein, 2024-11-02:
what about now, is it there now? cause i cant find it

closed 2024-11-04 by valeros