platformio/platform-espressif32 #918: esp32c3 - Invalid ARDUINO_PARTITION_ macro name

henrygab, 2022-09-20:
_Related to, but different from, #787 and #81._

## Issue
Build displays the following warning:
`<command-line>: warning: ISO C++11 requires whitespace after the macro name`

## Steps to Repro:
1. clone [WLED](https://github.com/Aircoookie/WLED.git) repository
2. Select the `esp32c3` environment
3. Build (using platformio)

## Expected Results:
Clean build

## Actual Results:
Compiler is passed invalid macro name related to ARDUINO_PARTITION_...

## Apparent Root Cause:
`-DARDUINO_PARTITION_tools/WLED_ESP32_4MB_1MB_FS`

Using verbose build, I was able to discover that the above define was being passed during compilation.  This is because the partition file was being stored in a subdirectory.

However, the `/` character is **_NOT_** a valid character for a define / macro name.  Therefore, the warning is accurate ... the parameter is invalid because it defines an invalid macro name.

Therefore, it appears there is a tooling problem, as it prevents partition information files from being stored in subdirectories.

Jason2866, 2022-09-24:
We (Tasmota) solved this by using our own boards manifests in folder `boards`.
In the boards manifests we define the partition scheme too. No problem to do it there.
[Example](https://github.com/arendst/Tasmota/blob/development/boards/esp32c3cdc.json)

valeros, 2022-09-27:
Hi @henrygab ! Apparently, you are using a completely custom development platform with third-party packages. Unfortunately, we cannot help you here as the issue was resolved some time and the latest official Arduino packages contain fixes for this problem. Please contact the maintainers of the third-party platform you're using in your project.

closed 2022-09-27 by valeros

henrygab, 2022-09-27:
My apologies!  I didn't realize it was using a custom BSP.  Thanks for confirming it's fixed in latest official Arduino packages.