platformio/platform-espressif32 #919: I upgraded the ESP32 PICO D4 firmware to  ESP32 PICO V3 ，Has been unable to start

K-LG, 2022-09-21:
It keeps printing the following in a loop
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

[image]
[image]
[image]

I hope you can help me. Thank you

K-LG, 2022-09-21:
I suspect that there is no bootloader.bin file for ESP32 PICO V3

Jason2866, 2022-09-21:
You use a old platform framework. Pico is not supported there. Use latest platform!

K-LG, 2022-09-21:
@Jason2866 Thanks for your reply,  I tried the 5.1.0 version and there were other problems

K-LG, 2022-09-21:
I experienced Guru Meditation Error after using 5.1.0:
[image]

K-LG, 2022-09-21:
There are flash frequency issues
I have tried to modify it, but I don't know whether it is normal
board_build.f_flash = 40000000L

Jason2866, 2022-09-21:
It is your sketch or a used library causing the problems. I have a board with a pico v3 which does work without issues.

valeros, 2022-10-04:
Hi @K-LG ! Does it work as expected in the Arduino IDE?

slaesh, 2022-10-28:
Guess [this](https://github.com/platformio/platform-espressif32/issues/947) is related?

closed 2022-12-16 by stale[bot]