platformio/platform-espressif32 #920: How to make to work ESP32-S3-MINI-1 with USB Host?

delpo, 2022-09-21:
I've been working with a ESP32-S3-WROOM-1 with no problems since latest version (5.1.1) was released.

But now I have to work with a ESP32-S3-MINI-1. The only way I could compile it was with this platformio.ini configuration:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```

But with this the USB Host is not available, I get this message when I try to compile:

```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.1.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20004.220825 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- EquipCaptivePortal-0.1.0
|   |-- ESPAsyncWebServer-master
|   |   |-- FS @ 2.0.0
|   |   |-- AsyncTCP-master
|   |   |-- WiFi @ 2.0.0
|   |-- AsyncTCP-master
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- Preferences @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- elapsedMillis-1.0.6
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\lib7fe\FS\FS.cpp.o
Compiling .pio\build\esp32dev\lib7fe\FS\vfs_api.cpp.o
Compiling .pio\build\esp32dev\libb17\AsyncTCP-master\AsyncTCP.cpp.o
Compiling .pio\build\esp32dev\lib326\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32dev\lib326\WiFi\WiFiAP.cpp.o
src/main.cpp:13:10: fatal error: usb/usb_host.h: No such file or directory
 #include <usb/usb_host.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
======================================================== [FAILED] Took 6.17 seconds ========================================================
```

It is planned to develop support for the MINI version? Any workaround I can do?

Jason2866, 2022-09-21:
Using a esp32 boards def will not work. If you submit a link to your ESP32-S3-MINI-1 where the specs can be seen a working config can be done.

delpo, 2022-09-21:
Thanks for your quick reply @Jason2866 ! This is the link of the documentation of the dev board with the microcontroller I am using:

ESP32-S3-DevKitM-1 https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitm-1.html

Jason2866, 2022-09-21:
Which memory variant do you have? Flash/PSRAM variant and size?

Jason2866, 2022-09-21:
Try this
```
[env:esp32s3-cdc]
platform = espressif32
board = adafruit_qtpy_esp32s3_nopsram
framework = arduino
```
the settings of this board does match yours

closed 2022-11-27 by stale[bot]