platformio/platform-espressif32 #923: Unable to compile 16MB partition > 3.4.0

othorne-public, 2022-09-27:
Hi, 

[I posted this on the Platformio forums and was instructed to post here](https://community.platformio.org/t/esp32-16mb-partition-framework-version-question/29799) 

I’m using an ESP32-Wroom-32E, 16MB version, I’ve been compiling with arduino version 3.4.0, which was the latest when I started my project.

After updating to any newer version (I haven’t tracked down the exact number it breaks on), any custom partition setup that totals over 4MB usage fails to run. The build starts, completes, uploads without any complaint, but simply will not run any code. This is reproduceable for me on any project, even a truly minimal one file project like this.

```
#include <Arduino.h>
void setup()
{
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println("Hello!");
}
void loop()
{
}
```

The terminal gets this far, but will not proceed any further. Reverting to an earlier framework gets it working again. I have LED’s attatched to my board that do not light up, so I presume my code isn’t running at all.

```
Writing at 0x000e669a... (100 %)
Wrote 897936 bytes (520603 compressed) at 0x00010000 in 13.0 seconds (effective 551.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================================================================== [SUCCESS] Took 57.44 seconds ==============================================================================================
--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
```

My ideal partition table looks like this.

partitions.csv

```
# Name    ,  Type  ,  SubType  ,  Offset  ,    Size    ,    Flags
nvs       ,  data  ,  nvs      ,  0x9000  ,    0x5000  ,
otadata   ,  data  ,  ota      ,          ,    0x2000  ,
ota_0     ,  app   ,  ota_0    ,          ,  0x400000  ,   
ota_1     ,  app   ,  ota_1    ,          ,  0x400000  ,   
littlefs  ,  data  ,  spiffs   ,          ,  0x7A4240  ,   
```

Before - Working

platformio.ino
```
[env:esp32dev]
board = esp32dev
framework = arduino
platform = https://github.com/platformio/platform-espressif32/archive/refs/tags/v3.4.0.zip
monitor_speed = 115200
monitor_port = COM7
upload_port = COM7
```

After - Not working

platformio.ino
```
[env:esp32dev]
board = esp32dev
framework = arduino
platform = https://github.com/platformio/platform-espressif32/archive/refs/tags/v5.1.1.zip
monitor_speed = 115200
monitor_port = COM7
upload_port = COM7
```

I've been unable to locate any other discussions, but I can reproduce this on any PC I've tried the project on so far. Is this a bug or have I misconfigured my platformio/partitions file?

Jason2866, 2022-09-27:
Your partition table is invalid. You can read this https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/tutorials/partition_table.html

othorne-public, 2022-09-27:
Appreciate the tip, but unfortunately that doesn't work for me either. I tried a few custom partition setups and the one you just supplied here, it still has the same behaviour. These files DO work on the older framework however.

```
Leaving...
Hard resetting via RTS pin...
======================================= [SUCCESS] Took 36.42 seconds =======================================--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
�
```

othorne-public, 2022-09-27:
I see, I'll run over that. For the moment, here's a minimal project, starts up, says "Hello!" does nothing else. If I uncomment partitions.csv or put in a 4MB table, it works, >4MB exhibits the issue. 

https://github.com/othorne-public/esp32_16mb_test

Jason2866, 2022-09-27:
Use this and it will start working:
```
[env:esp32dev]
framework = arduino
platform = espressif32 @ 5.1.1
board = esp32dev
board_build.f_cpu = 240000000L
board_build.f_flash = 40000000L
board_build.flash_mode = dio
board_build.flash_size = 16MB
board_build.partitions = partitions.csv
monitor_speed = 115200
```
```
.ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Hello!
```

othorne-public, 2022-09-27:
Cheers, Jason, I appreciate the help but it still won't budge. 
I've update the repo, your platformio file, verbatim works as intended until I put in larger partitions.

In case you're getting suspicious, this is 100% a 16MB module and I have the same issue on multiple computers.

Jason2866, 2022-09-27:
I did test the code and setup with my 16MB esp32 device. So only one option left. Your hardware is defect.

othorne-public, 2022-09-27:
I'm not completely convinced as the earlier version still works as expected. Could I be using an outdated/incompatible bootloader, maybe?

Jason2866, 2022-09-27:
You can try flash mode dout. There are a few early boards which work only with dout.
Bootloaders are verified working in release core 2.0.5. and the are used when you use the settings above. Maybe your platformio setup is screwed. You can delete the hidden folder `.platformio` with all stuff in there. Platformio will rebuild. So you start with a fresh install

Pixel-Soup, 2022-10-18:
I could not get my 16M Wrover to run with a 16M partition table until I added 2 extra lines to platformio.ini

board_build.partitions = default_16MB.csv
board_upload.flash_size =   16MB
board_upload.maximum_size = 16777216

closed 2022-12-03 by stale[bot]

timf34, 2024-06-02:
@Pixel-Soup thank you!!! Have spent too many hours trying to fix this