platformio/platform-espressif32 #925: UART Rx stalls after approx. 40 min of transfer

g3gg0, 2022-09-28:
Hi,

### Situation
I have built a WiFi <-> UART gateway using the ESP32.
Unfortunately after a 40 minute transfer period, the Rx path of the ESP32 stalls and no activity on UART is "seen" anymore.
If you then wait 40 minutes again, the data is suddenly seen again and the ESP32 works as before.

[40 min working] [40 min no Rx] [40 min working] ....

Tx is working as intended, the connected external peripheral is receiving the data as before.

### Details
First I used the arduino Serial implementation for Rx/Tx, but switched to the underlying IDF as a measure to overcome this problem.
But the problem persisted.
This problem is known to me for over a year now, not seeing any fix so far.
[image]
https://github.com/g3gg0/ESP232/blob/loopback_test/src/Serial.ino#L200
There is the code which reads data from the UART buffers, both Serial.available() and uart_read_bytes() return zero during that "no Rx" period.

### Deep dive
Now I digged into debugging, first by investigating the UART peripheral registers.

Status "Rx/Tx are working properly"
[image]

Status: "Tx working, Rx stalled"
[image]

There I could see, that the UART_INT_ENA_REG shows disabled Rx interrupts, but even when forcefully enabling again using a telnet based debug interface, they were disabled again instantly.
In the "no Rx" period, the UART_STATUS_REG however shows the RXFIFO_CNT field is counting continuously as opposed to the "working" period where it stays virtually zero.
[image]

### This leads me to my personal conclusion:
### The underlying ESP-IDF has a bug, causing the Rx path to maybe overflow some counters, queues or whatever.

### Test case
I modified my project to "self-serve" the communication using UART1 in the branch loopback_test
And guess what, it also stalls after approx. that period.
Sources: https://github.com/g3gg0/ESP232/tree/loopback_test

Just build and flash that project. The first time the SPIFFS will format, but after that you should see some communication like that:

SCPI communication (simplified):
```
> *OPC?
< 1
> *CLS
> DISP:WIND1:DATA?
< " -0.0000 A"
> *STB?
< 1
```
Repeating 4 times a second.

Screenshot in action:
[image]

after the 40 minutes, the communication would look like this:
[image]

And this whole process will repeat over and over.

If you want to break into it, the lines 92, 129 and 165 will give you a good spot 
https://github.com/g3gg0/ESP232/blob/loopback_test/src/SCPI.ino#L129

Regards,
g3gg0

Jason2866, 2022-09-29:
This is a issue from espressif Arduino core. This is IDE Platformio. There is no bug.

g3gg0, 2022-09-29:
Hi Jason,

maybe i am wrong, but this project "platform-espressif32" is the provider for the espressif-idf along with required markup.

As it is using some (to me unclear) version of the IDF, this is my first obvious contact point.
I don't know if the project use a more recent version of the IDF, or if there is already a branch.

The Espressif IDF project would point me into using a more recent version first, which I would again address here.

Regards,
Georg

g3gg0, 2022-09-29:
Sorry, that this looks a bit intransparent to me.

Do you mean https://github.com/espressif/arduino-esp32 is providing the package?
However this project uses the IDF directly.
(via https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py if i am not wrong)

Jason2866, 2022-09-29:
Latest release from espressif is IDF 4.4.2 which is used from Platformio. If you want to use a unstable devekopment version you walk alone.

g3gg0, 2022-09-29:
Ah your post grew a bit since the initial reply :D

Thanks for your reply! I currently am testing your branch, but as it uses 4.4 instead of 4.4.2, I expect the bug to be still persistent. Have seen this ages ago and waited for a fix till then.
Except if you backported the mentioned serial fixes(?)
Will keep you updated if it also appears.

I also ran into a few issues of this kind in the past, even with the peripherals. Especially RMT annoyed me a bit.

Parallel, I am filing an issue in esp-idf. Don't want to implement a serial driver on my own.

Regarding hardware: thanks :) will probably switch to a PICO series to make it disappear in a plastic D-SUB housing.

Regarding tasmota: hmm didn't think of this to be honest.
https://tasmota.github.io/docs/Serial-to-TCP-Bridge/
If you say that the serial fight is already fought and it is expected to be stable, then the only thing left would
be the small SCPI modification and publishing the results via MQTT.
Good point, will definitely think about that!

Jason2866, 2022-09-29:
My build uses IDF 4.4.3. Maybe we see a PR for Tasmota.

g3gg0, 2022-09-29:
I see, thanks.

g3gg0, 2022-09-29:
IDF 4.4.3 shows the same behavior :(

valeros, 2022-10-04:
Hi @g3gg0, have you tried using the official ESP-IDF development environment (e.g. for [VS Code](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md))? Does it work there?

g3gg0, 2022-10-04:
hi, sorry for not updating here. it figures out, this exact failure is totally my fault.
the udp debug code was implemented to track the described problem, but it was not as consistent as it was here.
maybe it was already closed, but i was distacted by my own bad code :-(

closed 2022-10-04 by g3gg0