platformio/platform-espressif32 #926: Cannot compile VL6180X sensor library with ESP32-S3-M

lsolsona, 2022-09-29:
I am using a ESP32-S3-DevKitM-1 with PlatformIO and Arduino Framework. This is my platformio.ini config:

```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

; change MCU frequency
board_build.f_cpu = 240000000L
```

When I try to use VL6180X sensor (https://www.sparkfun.com/products/12785) libraries I get compile errors. I have tried with two libraries: 

- https://github.com/sparkfun/ToF_Range_Finder_Sensor-VL6180 (my first choice)
- https://github.com/DFRobot/DFRobot_VL6180X (an alternative)

This is the compile error I get with Sparkfun library (same error with the other one):
```

*  Ejecutando tarea: C:\.platformio\penv\Scripts\platformio.exe run 

Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
-----------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (5.1.1) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20004.220825 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- EquipMotor-1.0.0
|-- EquipSerialCom-1.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|-- rfid-1.4.10
|   |-- SPI @ 2.0.0
|-- EquipSickGTB6-1.0.0
|-- SX1509 IO Expander @ 3.0.4
|   |-- Wire @ 2.0.0
|-- SparkFun_ToF_Range_Finder-VL6180_Arduino_Library-1.1.3
|   |-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- Stepper @ 1.1.3
|-- Wire @ 2.0.0
Building in release mode
Linking .pio\build\esp32-s3-devkitc-1\firmware.elf
c:/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: cannot open map file C:/Users/LucíaSolsonaMassó/Documents/EAGLE/projects/equipo_pya_automatico/code/equipo_pya_automatico_MCU2/.pio/build/esp32-s3-devkitc-1/firmware.map: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1
======= [FAILED] Took 10.13 seconds =======
```

The same first library (the Sparkfun one) works with other microcontroller (ESP32-DevKitM-1), so I believe it is a Arduino Framework problem. Any idea?

valeros, 2022-09-29:
Seems like your username `LucíaSolsonaMassó` may cause the problem. Try creating a new project somewhere in the root of your file system.

lsolsona, 2022-10-21:
Sorry for the late response. Seems like @valeros was right, it was a problem with my user name. It's strange because I always put all my projects in that route and never had a problem like this. Moreover, the problem sometimes vanished when I commented some of my `#includes` randomly.

Nevertheless, changing the project to other place or changing muy user name solved the problem. Thank you!

closed 2022-10-21 by lsolsona