platformio/platform-espressif32 #927: Feature request: Using local copy of arduino-esp32 repository/component

dmkitsch, 2022-09-30:
I need to create a custom configuration and build of the arduino esp32 platform that includes roaming support. Is there any way I can use this build locally within platformio using this espressif32 platform component?

Thanks

Jason2866, 2022-09-30:
Here is a example for using Arduino as a component with IDF. https://github.com/Jason2866/LILYGO-T5-4.7-E-Paper_Weather-Station
You can configure sdkconfig for your needs.

valeros, 2022-09-30:
Hi @dmkitsch, the following pages may be helpful:
https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder
https://docs.platformio.org/en/latest/scripting/examples/override_package_files.html

closed 2022-09-30 by valeros