platformio/platform-espressif32 #928: dfrobot_firebeetle2_esp32s3: Wrong --chip Argument

grwells, 2022-10-01:
Problem: uploading to DFRobot Firebeetle ESP32-E fails with error: `A fatal error occurred: This chip is ESP32 not ESP32-S3. Wrong --chip argument?`

This problem can be overcome two ways:
1. Build with espressif version 5.1.1 and specifying `firebeetle32` as board.
2. Building with the `platformio.ini` below and specifying `board_build.mcu = esp32`.

```
[env:firebeetle32]
; board definition incorrect for firebeetle32-e as of espressif 5.1.1
platform = espressif32@^5.1.1
board = dfrobot_firebeetle2_esp32s3
;board_build.mcu = esp32
framework = arduino
build_type = debug
build_flags = -DDEBUG_ESP_HTTP_UPDATE, -DDEBUG_ESP_PORT=Serial
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
board_build.partitions = max_ota.csv
lib_deps = 
		bblanchon/ArduinoJson@^6.19.4
	   	knolleary/PubSubClient
```

Build Platform and Packages

```
PLATFORM: Espressif 32 (5.2.0) > DFRobot Firebeetle 2 ESP32-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
```

I believe this is a simple mistake, and easily fixed. Thankful that the spec differences between `firebeetle32` and version 2 are being updated in pio!

PlatformIO Core, version 6.1.4
Operating System: Pop!_OS

grwells, 2022-10-01:
The second solution I posted is not a full solution as hoped. It fixes the upload errors but somehow broke code in my project that relies on I2C... not sure what the real problem is but building for `esp-wrover-kit` fixes both issues.

Jason2866, 2022-10-01:
Simple solution, this tells what you have `A fatal error occurred: This chip is ESP32 not ESP32-S3. Wrong --chip argument?`
Your board is a ESP32 **not** a ESP32-S3. You have to use the correct boards.json probably firebeetle32

grwells, 2022-10-01:
# 
> Your board is a ESP32 not a ESP32-S3. 

I have the [DFRobot Firebeetle 2 ESP32-E](https://github.com/platformio/platform-espressif32/commit/4d0b78a9cb69a8bf0f38c2252e2841ac249f2290), so I would tend to agree with your statement. The real problem here is that there hasn't been a board definition for the Firebeetle32 v1 AND v2, which I have, until 2 months ago. When the definition for v2 was added, it was added as an ESP32-S3. I don't see any explicit proof that what I have is an ESP32-S3 from the vendor or the packaging. 

Until a week ago I was compiling with `board = firebeetle32` because although DFRobot had released both versions of the board, that change had not made its way into platformio. As of [this commit](https://github.com/platformio/platform-espressif32/commit/4d0b78a9cb69a8bf0f38c2252e2841ac249f2290) the two iterations of the Firebeetle ESP32 were given their own boards.json specs, but spec for the board I have results in using incorrect build arguments.

#

> You have to use the correct boards.json probably firebeetle32

It might actually be simpler to use `firebeetle32.json`, but since this is likely to be a problem for others and generally confusing I would like to help find a solution for `dfrobot_firebeetle2_esp32s3.json`. I am hoping someone who was part of the decision to specify the Firebeetle 2 ESP32-E as an ESP32-S3 would be able to weigh in on why that needed to happen and how the build config can be modified to work with the physical board.

Jason2866, 2022-10-01:
`dfrobot_firebeetle2_esp32s3` clearly has S3 in name. So it is for S3.
It is a fact that your board is an ESP32 since it is identified from esptool.py with `A fatal error occurred: This chip is ESP32 not ESP32-S3. Wrong --chip argument?`
So the actual boards manifest are all correct. For crude naming only Dfrobot is to blame.

grwells, 2022-10-01:
@Jason2866 are you saying that DFRobot wrote `dfrobot_firebeetle2_esp32s3.json`? Because although I agree that the naming is crude and an error, it doesn't change the fact that the "manifest" doesn't work (ergo is **not** correct) for the physical board and is thus worthless until fixed. 

So who fixes it? Should I go contact DFRobot or just start hacking the manifest until it works and then propose a change?

grwells, 2022-10-01:
Maybe part of the problem is that I don't understand the process for how board manifests are proposed and then adopted by PlatformIO. Or who assumes responsibility for amending them during the time that the board is supported for that matter.

Jason2866, 2022-10-01:
Platformio crew usually takes the boards from the Arduino boards.txt or via PRs which have been contributed. So first step is to check if the error is in Arduino too.

grwells, 2022-10-01:
Okay, good to know. I will go check if building in ArduinoIDE makes any difference. It may work since that PR doesn't seem to classify the board as an ESP32-S3 anywhere besides the name of the board definition.

Jason2866, 2022-10-01:
It clearly is a S3 board in Arduino boards.txt!

Jason2866, 2022-10-01:
Platformio boards manifest is 10ü% correct.

grwells, 2022-10-01:
> Platformio boards manifest is 100% correct.

Haha, that may be true, since apparently it is just an adoption of the Arduino boards.txt. I will need to see it work properly on the actual board before I believe it though.

I appreciate the insight into the process, I will go see if I can pick apart what specifically is breaking in my project for more clues. I know it is something in the I2C operations and the fact that the physical chip is esp32 not s3.

Jason2866, 2022-10-01:
Do NOT use this boards manifest for your ESP32. NEVER use a boards manifest from a not matching MCU. You can use many ESP32 boards manifest and trim it to your needs.
MCU have to match in every case!

valeros, 2022-10-04:
Please use appropriate board ID that corresponds to your hardware.

closed 2022-10-04 by valeros