platformio/platform-espressif32 #932: assert failed: twai_handle_tx_buffer_frame twai.c:183 (p_twai_obj->tx_msg_count >= 0) if CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=y (IDFGH-8204)

EmbeddedDevver, 2022-10-04:
The TWAI interface has a bug that causes the chip to reboot if you short / have interference the CANBUS pins. There is a proposed solution in the [IDF](https://github.com/espressif/esp-idf/issues/9697 ) github.

For that I need to adjust the IDF TWAI component:

How can I configure the IDF configuration in the platformIO enviroment?

This issue is identical to this bug:

Jason2866, 2022-10-04:
No you have to configure
```
# CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST is not set
```
to avoid crashing until the bug is fixed in IDF. In the meantime
- use Arduino core 2.0.4 (this change was made after release of core 2.0.4)
- or compile you Arduino libs yourself. See https://github.com/espressif/esp32-arduino-lib-builder
- use IDF with Arduino as Component

EmbeddedDevver, 2022-10-04:
Thanks Jason for your reply. 

How can I compile the TWAI for platformIO myself, I want to fix it for (platform = espressif32 @ ^4.4.0) ?  Do you know where I can start with that?
Thank you!

I use platform io

When using 

```
platform = espressif32 @ ^4.4.0
```

It uses TWAI and then I have this bug

When using

```
platform = espressif32 @ ^3.5.0
```
Then there is no problem.

Jason2866, 2022-10-04:
Than it is not the setting mentioned. This was changed lately. https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

EmbeddedDevver, 2022-10-04:
Please note
espressif32 3v5 does not use TWAI but CAN.
espressif32 4.4 uses TWAI (with the bug)

Jason2866, 2022-10-04:
Not a platformio issue anyways. You can start here https://community.platformio.org/t/framework-espidf-arduino/14949

closed 2022-10-04 by EmbeddedDevver

f-hoepfinger-hr-agrartechnik, 2023-01-16:
espressif/esp-idf#9697