platformio/platform-espressif32 #933: *** [.pio\build\Pico-v3-02\partitions.bin] Source `default_8M.csv' not found

murarduino, 2022-10-04:
Vscode espressif32 v5.2.0 today updateed.
MCU：PICO-V3-02 
I try used custom partitions cvs file or existing file to building ,but it's error.
partitions files directory :
..\.platformio\packages\framework-arduinoespressif32\tools\partitions

murarduino, 2022-10-04:
platfromio.ini:
[env:Pico-v3-02]
platform = espressif32
board = denky_d4
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.partitions = default_8M.csv
lib_deps = esphome/ESP32-audioI2S@^2.1.0

valeros, 2022-10-04:
Hi @murarduino ! If you are using partition files from the framework, then it should be `default_8MB.csv`, not `default_8M.csv`.

closed 2022-10-04 by valeros

murarduino, 2022-10-04:
> Hi @murarduino ! If you are using partition files from the framework, then it should be `default_8MB.csv`, not `default_8M.csv`.
Thank you, it was my negligence that I didn't write all the file names.
BTW, I would like to ask, is the directory of \packages\framework-arduinoespressif32@XXXX still useful?

valeros, 2022-10-04:
> BTW, I would like to ask, is the directory of \packages\framework-arduinoespressif32@XXXX still useful?

It can be, for example, if some of your projects use a specific version of dev-platform, e.g. `platform = espressif32 @ ^ 5.1.0`