platformio/platform-espressif32 #936: Build failed with esp-matter light

Diegorro98, 2022-10-09:
I'm trying to build the light example in order to help contributing to #854 now that Matter has achieved v1.0.0 release.
To build with platformio I did the following on the esp-matter light example:

- I created a `platform.ini` with the following content:
```
[platformio]
src_dir = main

[env]
platform = espressif32
framework = espidf
monitor_speed = 115200
extra_scripts = pre:pre.py

[env:esp32dev]
board = esp32dev
```

- Created an extra-script that will be executed before building (as declared at platform.ini) to solve some environment variables problems:
```
import os

os.environ["PATH"] += ":/workspaces/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/"
os.environ["ESP_MATTER_PATH"] = "/workspaces/esp-matter"
```

But build failed and this showed at the build log:
```
*** Two environments with different actions were specified for the same target: /workspaces/esp-matter/examples/light/.pio/build/esp32dev/chip.o
(action 1: xtensa-esp32-elf-g++ -o chip.o -c -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/workspaces/esp-matter/examples/light=. -fmacro-prefix-map=/home/codespace/.platformio/packages/framework-espidf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -Wno-error=uninitialized -Wno-error=maybe-uninitialized -Wno-missing-field-initializers -Wno-format-nonliteral -Wno-format-security -Wformat=0 -Wno-old-style-declaration -Os -DPLATFORMIO=60104 -DARDUINO_ESP32_DEV -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -D_GNU_SOURCE -DIDF_VER=\"4.4.2\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS chip.cpp)
(action 2: xtensa-esp32-elf-g++ -o chip.o -c -std=gnu++11 -std=gnu++14 -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/workspaces/esp-matter/examples/light=. -fmacro-prefix-map=/home/codespace/.platformio/packages/framework-espidf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -Wno-error=uninitialized -Wno-error=maybe-uninitialized -Wno-missing-field-initializers -Wno-format-nonliteral -Wno-format-security -Wformat=0 -fno-exceptions -fno-rtti -Os -DPLATFORMIO=60104 -DARDUINO_ESP32_DEV -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DCHIP_HAVE_CONFIG_H -D_GNU_SOURCE -DIDF_VER=\"4.4.2\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS chip.cpp)
File "/home/codespace/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 643, in compile_source_files
```
I tried to solve it by setting build flags (and unflags) to `platformio.ini`:
```
build_unflags = 
	-std=gnu++11
	-Wno-old-style-declaration
build_flags =
	-std=gnu++14
	-DCHIP_HAVE_CONFIG_H
	-fno-rtti
	-fno-exceptions
```
But then I got this:
```
warning: Two different environments were specified for target /workspaces/esp-matter/examples/light/.pio/build/esp32dev/chip.o,
        but they appear to have the same action: ${TEMPFILE('$CXX -o $TARGET -c $CXXFLAGS $CCFLAGS $_CCCOMCOM $SOURCES','$CXXCOMSTR')}
File "/home/codespace/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 643, in compile_source_files

*** Multiple ways to build the same target were specified for: /workspaces/esp-matter/examples/light/.pio/build/esp32dev/chip.o  (from ['/workspaces/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/chip.c'] and from ['/workspaces/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/chip.cpp'])
File "/home/codespace/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 643, in compile_source_files
```
To avoid more problems with `chip.o`, I deleted `chip.c` (for example, maybe `chip.cpp` can be deleted instead too) and references to this file because I didn't know how to continue, but build failed again:
```
Compiling .pio/build/esp32dev/main/app_driver.o
sh: 1: cannot open lib/address_resolve/AddressResolve_DefaultImpl.h: No such file
Compiling .pio/build/esp32dev/main/app_main.o
Generating LD script .pio/build/esp32dev/memory.ld
sh: 1: cannot open lib/address_resolve/AddressResolve_DefaultImpl.h: No such file
Compiling .pio/build/esp32dev/app_ble.o
*** [.pio/build/esp32dev/main/app_driver.o] Error 2
*** [.pio/build/esp32dev/main/app_main.o] Error 2
```
The file refers to `/workspaces/esp-matter/connectedhomeip/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.h` which exist...

I would be very grateful if anyone can help me to build the example

Jason2866, 2022-10-09:
With a quick look, i think you can not do this way. You are mixing to build environments.

Diegorro98, 2022-10-11:
So, any suggestion to achieve building this?

jakubdybczak, 2022-10-16:
Regarding `AddressResolve_DefaultImpl.h`, I'm not sure how did you tried to launch it, but I come across same problem, when trying to launch esp-matter with Platform IO. In my project I tried to add matter as IDF-component, so I guess it's another case but anyway maybe it will help you.

Please try adding escaping to esp_matter `CMakeLists.txt` so that last line looks like this:
`"-DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\\<lib/address_resolve/AddressResolve_DefaultImpl.h\\>"`

Btw. please let me know if you suceed with launching Matter with PIO. After the problem that you mentioned there are ton of more problems. Like for example CHIP does build some generated artifacts (AppBuildConfig.h / CHIPDeviceBuildConfig.h / SystemBuildConfig.h / ...) via `externalproject_add` instructions in cmake files, which aren't invoked in PIO environment. I don't have much knowledge about the stuff that is going on here but it looks like building pipeline is vastly different here than when running esp-idf's `idf.py build`.

valeros, 2022-10-17:
Might be a bug, can someone prepare a minimal PlatformIO project to reproduce the issue?

Diegorro98, 2022-10-17:
@jakubdybczak How should `CMakeList.txt` look like? because I tried to put literally that line at the end of the file, but there was a parse error, then I tried to add it to `COMPILE_OPTIONS` so last line looks like this:
```
idf_build_set_property(COMPILE_OPTIONS "-Wno-format-nonliteral;-Wno-format-security;-Wformat=0; -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\\<lib/address_resolve/AddressResolve_DefaultImpl.h\\>" APPEND)
```
But doesn't affect the result, I mean, I have same last error than before.
I would like to know if you have encountered same errors than me at the start before `AddressResolve_DefaultImpl.h` error.

@valeros I can prepare a PlatformIO project that contains esp-matter as git submodule, and I would create a `init.sh` file in order to download esp-matter, Matter, and the esp32 platform required submodules (so project size is lighter on disk). Would that be useful for you?
If the answer is yes, would you prefer the project at the starting point so you can reproduce all bugs I had, or do you prefer the version where I couldn't do anything else?

Diegorro98, 2022-11-26:
@valeros Did you get time to reproduce the issue?

Mem21, 2023-01-06:
I am trying to build esp-matter in platformio too. 
You need to use Espressif 32 5.3.0 that has esp-idf 4.4.3.

My build failed because the files mensioned by @jakubdybczak are missing.
I dont know how can i generate this files. Can someone help with this?

S-A-M-J, 2023-10-09:
Did anybody figure out how to get esp-matter running in platformIO and can perhaps give a quick explanation of how? I'm stumbling from one problem to another with no solution in sight.