platformio/platform-espressif32 #940: ESP32 ULP Build does not follow CMakeLists

DocWibbleyWobbley, 2022-10-13:
When building a project the ULP program is always built from all `.S` files in `ulp` folder and is always named `ulp_main`.
When building the same project with IDF-ESP the CMakeLists.txt is obeyed properly.
Therefore it's not possible to configure different ULP programs based on component configuration.
More info in this [PlatformIO discussion](https://community.platformio.org/t/esp32-ulp-build-does-not-follow-cmakelists/29987)

Jason2866, 2022-10-15:
The behaviour is as designed and as described in Platformio documentation.

DocWibbleyWobbley, 2022-10-17:
Can you provide a link to said documentation?
I seem to be unable to find anything about that.
I tried with the [Registry](https://registry.platformio.org/platforms/platformio/espressif32), [Documentation](https://docs.platformio.org/en/latest/platforms/espressif32.html), and looked in both ulp examples.

valeros, 2022-10-17:
Hi @DocWibbleyWobbley ! @Jason2866 is right, although it's not documented. Currently, all source files are collected from the `ulp` folder, mainly because the underlying IDF build system doesn't export necessary information to reproduce the build logic to the full extent.

closed 2022-10-17 by valeros

Jason2866, 2022-10-17:
The structure is described https://docs.platformio.org/en/latest/frameworks/espidf.html#ulp-coprocessor-programming

DocWibbleyWobbley, 2022-10-17:
@valeros I see
@Jason2866 Thanks

Jason2866, 2026-02-23:
@m-mcgowan  https://github.com/m-mcgowan/pio-ulp-cmake  -> 404 ?!