platformio/platform-espressif32 #942: Error creating new project in VSC + PlatformIO + IDF Framework

gerry62, 2022-10-13:
I'm trying to create a new project using the IDF framework. I get an error: "Could not initialize project"
With this output:

__PIO Core Call Error: "Resolving ttgo-lora32-v1 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...

Error: Processing ttgo-lora32-v1 (platform: espressif32; board: ttgo-lora32-v1; framework: espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v1.html
PLATFORM: Espressif 32 (5.2.0) > TTGO LoRa32-OLED V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, 
   minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
 Error: Detected a whitespace character in project paths.
 ========================== [FAILED] Took 0.44 seconds ==========================
 
 Environment Status Duration
 -------------- -------- ------------
 ttgo-lora32-v1 FAILED 00:00:00.440
 ==================== 1 failed, 0 succeeded in 00:00:00.440 ====================

I've searched for this error, and I've tried to create ./platformio directory in root. This did not help.
My username does not contain spaces and there are no spaces in my directory structure.
I'm working on an M1 Macbook Pro.
I've created various ESP32 projects with the Arduino framework, without encountering problems.
How do I get rid of this error?

Thanks

Jason2866, 2022-10-15:
Tried with my Mac M1 Air. Does work. You can try to delete the hidden folder `.platformio`
Platformio will rebuild. Maybe something has gone to the south.

gerry62, 2022-10-16:
@Jason2866 
Thanks for your reply!
Unfortunately, it did not fix the problem.
I solved it by disabling Dropbox sync. It seems Dropbox does some funny tricks to the filesystem when syncing, and a space character was inserted into a directory path.

closed 2022-10-16 by gerry62