platformio/platform-espressif32 #944: ESP32 S3 can't boot after update to 5.2.0

MassimilianoMontagni, 2022-10-17:
After updating to 5.2.0 from 5.1.1 my ESP32-S3 (N4R8 and N8R8) can't boot with the config that I'll put down here. When I downgrade by writing `platform = espressif32@5.1.1` the project restart to work.
I already did a flash erase before flashing the firmware.
The strange thing is, with 5.2.0, I can enter in download mode but when I reset the ESP32-S3 stays completely in silence without print anything to UART0. 
It's maybe a problem with the bootloader?

[sdkconfig.esp32-s3-devkitc-1.txt](https://github.com/platformio/platform-espressif32/files/9802218/sdkconfig.esp32-s3-devkitc-1.txt)
[platformio.ini.txt](https://github.com/platformio/platform-espressif32/files/9802222/platformio.ini.txt)
[partitionTable.csv](https://github.com/platformio/platform-espressif32/files/9802211/partitionTable.csv)

Jason2866, 2022-10-17:
Not seeing anything in serial console is in my experience a misconfiguration.
Nearly any wrong setting can end up in this behaviour.
So double check, flash size, flash types for flash / psram and partition table.

MassimilianoMontagni, 2022-10-17:
I already double checked the configuration (in fact with version 5.1.1 and priors is working) and I also checked the pins (IO46, IO0, UART0TX, UART0RX). The partition table is correct because I didn't changed anything and I also tried with the default partition table with OTA, and I tried even to disable PSRAM but still nothing.
If I create a new project with 5.2.0 is working so it must be something in the config (that I have attached) but I don't what

SopraniDaniele23, 2022-10-19:
I'm experiencing the same issue with a blank project with the same configuration and partitions as @MassimilianoMontagni

gohm44, 2022-10-20:
I had a similar issue. I solved it by revert  https://github.com/platformio/platform-espressif32/commit/940038601f1b690a20c43eb23cf8f119970ff7d9
`board_build.flash_mode= dio`

MassimilianoMontagni, 2022-10-20:
@gohm44 yes it works, thanks. I was only editing the sdkconfig and not the platformio.ini

closed 2022-10-20 by MassimilianoMontagni