platformio/platform-espressif32 #945: Using arduino as component in the espidf-peripherals-usb example

xtrinch, 2022-10-23:
Hello,

I've been wondering if it's at all possible to use arduino as component when using tinyusb.

I've tried adding arduino to https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-peripherals-usb,
but the project does not compile and fails with a bunch of errors. (And yes it compiles fine without adding arduino as component!)

It would be really nice to have an example project that does both usb and arduino as component.

Config:
```
[env]
platform = espressif32
framework = espidf, arduino
monitor_speed = 115200

[env:esp32-s2-saola-1]
board = esp32-s2-saola-1
```

Errors:

```
Compiling .pio/build/esp32-s2-saola-1/wiring_pulse.o
Compiling .pio/build/esp32-s2-saola-1/wiring_shift.o
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp: In function 'uint16_t load_cdc_descriptor(uint8_t*, uint8_t*)':
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:29:25: error: 'tinyusb_add_string_descriptor' was not declared in this scope
     uint8_t str_index = tinyusb_add_string_descriptor("TinyUSB CDC");
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32-s2-saola-1/WMath.o
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:30:24: error: 'TUD_CDC_DESC_LEN' was not declared in this scope
     uint8_t descriptor[TUD_CDC_DESC_LEN] = {
                        ^~~~~~~~~~~~~~~~
Compiling .pio/build/esp32-s2-saola-1/WString.o
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:32:13: error: 'TUD_CDC_DESCRIPTOR' was not declared in this scope
             TUD_CDC_DESCRIPTOR(*itf, str_index, 0x85, 64, 0x03, 0x84, 64)
             ^~~~~~~~~~~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:35:17: error: 'descriptor' was not declared in this scope
     memcpy(dst, descriptor, TUD_CDC_DESC_LEN);
                 ^~~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:35:17: note: suggested alternative: 'encrypt'
     memcpy(dst, descriptor, TUD_CDC_DESC_LEN);
                 ^~~~~~~~~~
                 encrypt
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp: At global scope:
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:48:42: error: 'cdc_line_coding_t' has not been declared
 void tud_cdc_line_coding_cb(uint8_t itf, cdc_line_coding_t const* p_line_coding)
                                          ^~~~~~~~~~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp: In function 'void tud_cdc_line_coding_cb(uint8_t, const int*)':
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:51:52: error: request for member 'bit_rate' in '* p_line_coding', which is of non-class type 'const int'
         devices[itf]->_onLineCoding(p_line_coding->bit_rate, p_line_coding->stop_bits, p_line_coding->parity, p_line_coding->data_bits);
                                                    ^~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:51:77: error: request for member 'stop_bits' in '* p_line_coding', which is of non-class type 'const int'
         devices[itf]->_onLineCoding(p_line_coding->bit_rate, p_line_coding->stop_bits, p_line_coding->parity, p_line_coding->data_bits);
                                                                             ^~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:51:103: error: request for member 'parity' in '* p_line_coding', which is of non-class type 'const int'
         devices[itf]->_onLineCoding(p_line_coding->bit_rate, p_line_coding->stop_bits, p_line_coding->parity, p_line_coding->data_bits);
                                                                                                       ^~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:51:126: error: request for member 'data_bits' in '* p_line_coding', which is of non-class type 'const int'
         devices[itf]->_onLineCoding(p_line_coding->bit_rate, p_line_coding->stop_bits, p_line_coding->parity, p_line_coding->data_bits);
                                                                                                                              ^~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp: In constructor 'USBCDC::USBCDC(uint8_t)':
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:99:30: error: 'USB_INTERFACE_CDC' was not declared in this scope
     tinyusb_enable_interface(USB_INTERFACE_CDC, TUD_CDC_DESC_LEN, load_cdc_descriptor);
                              ^~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32-s2-saola-1/ArduinoOTA.o
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:99:49: error: 'TUD_CDC_DESC_LEN' was not declared in this scope
     tinyusb_enable_interface(USB_INTERFACE_CDC, TUD_CDC_DESC_LEN, load_cdc_descriptor);
                                                 ^~~~~~~~~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:99:5: error: 'tinyusb_enable_interface' was not declared in this scope
[... 93 lines cut ...]
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:390:13: note: suggested alternative: 'tud_cdc_tx_complete_cb'
         if(!tud_cdc_n_connected(itf)){
             ^~~~~~~~~~~~~~~~~~~
             tud_cdc_tx_complete_cb
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:394:24: error: 'tud_cdc_n_write_available' was not declared in this scope
         size_t space = tud_cdc_n_write_available(itf);
                        ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:396:13: error: 'tud_cdc_n_write_flush' was not declared in this scope
             tud_cdc_n_write_flush(itf);
             ^~~~~~~~~~~~~~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:402:23: error: 'tud_cdc_n_write' was not declared in this scope
         size_t sent = tud_cdc_n_write(itf, buffer+so_far, space);
                       ^~~~~~~~~~~~~~~
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:402:23: note: suggested alternative: 'tud_cdc_rx_cb'
         size_t sent = tud_cdc_n_write(itf, buffer+so_far, space);
                       ^~~~~~~~~~~~~~~
                       tud_cdc_rx_cb
/home/xtrinch/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBCDC.cpp:406:13: error: 'tud_cdc_n_write_flush' was not declared in this scope
             tud_cdc_n_write_flush(itf);
             ^~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32-s2-saola-1/EEPROM.o
Compiling .pio/build/esp32-s2-saola-1/ESPmDNS.o
Compiling .pio/build/esp32-s2-saola-1/ETH.o
*** [.pio/build/esp32-s2-saola-1/USBCDC.o] Error 1
```

Jason2866, 2022-10-23:
You have to clone (**not** recursive!) this repo https://github.com/hathach/tinyusb in Arduino .../libraries/
to get a working Arduino tinyusb

xtrinch, 2022-10-24:
Could you please be more specific as to where I need to clone tinyusb to? What is the path of “.../components“?

Jason2866, 2022-10-24:
Platformio does not install "Arduino Lib Builder".  The libs for the Arduino framework are compiled with https://github.com/espressif/esp32-arduino-lib-builder With this script https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/update-components.sh the needed additional components needed for Arduino are added to compile the complete Arduino components with IDF.

xtrinch, 2022-10-24:
So I compiled with the lib builder, and copied `components/arduino` and `components/arduino_tinyusb` into the project directory, but the error messages remain the same. I'm a tad confused because I never saw the actual project having the `components` folder with platformio - isn't this for building outside platformio?

closed 2022-12-03 by stale[bot]