platformio/platform-espressif32 #946: Adding for Heltec V3 Boards

Baptou88, 2022-10-26:
Hello everyone, 

the Heltec company has developed a new product range which is an evolution of its old product.
the update concerns these boards:
*[Wifi Lora 32 V3](https://heltec.org/project/wifi-lora-32-v3/)
*[Wireless Stick Lite (V3)](https://heltec.org/project/wireless-stick-lite-v2/)
*[Wireless Shell (V3)](https://heltec.org/project/wireless-shell-v3/)
*[WiFi Kit 32 (V3)](https://heltec.org/project/wifi-kit-32-v3/)

can you please update your repository , so that we can work on these new boards with platformio

smerkadidit, 2023-01-19:
This would be great... im running into issues with this

valeros, 2023-03-09:
The [Heltec WiFi LoRa 32 (V3)](https://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_lora_32_V3.html) and [Heltec WiFi Kit 32 (V3)](https://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_kit_32_V3.html) already available in the [latest release](https://github.com/platformio/platform-espressif32/releases/tag/v6.1.0). AFAIK, Wireless shell(V3) and Wireless stick lite (V3) are similar to Heltec WiFi Kit 32 (V3) so you can use the `heltec_wifi_lora_32_V3` ID.

closed 2023-03-09 by valeros

LenStruttmann, 2023-03-10:
Newbie question: I just now tried to create a project, selecting board "Heltec Wifi Kit 32 (V3)" in the drop-down list and I get:

`PIO Core Call Error: "The current working directory ...
...Resolving heltec_wifi_kit_32_V3 dependencies...\r\n\n\nError: Unknown board ID 'heltec_wifi_kit_32_V3'"`

How do I update Platformio to see this board?

Thanks.

mistrjirka, 2024-03-01:
What wireless stick lite v3? what should I use?

Baptou88, 2024-03-02:
hello @mistrjirka,
if you use this chip: [Wireless Stick lite V3](https://heltec.org/project/wireless-stick-lite-v2/), my platformio config file looks like :
``` ini
[env:heltec_wifi_lora_32_V3]
board = heltec_wifi_lora_32_V3
framework = arduino

```

BrentonPoke, 2024-11-04:
```
======================================================================================================================================================================
ID                                   MCU      Frequency    Flash    RAM     Name

heltec_wireless_stick                ESP32    240MHz       8MB      320KB   Heltec Wireless Stick
heltec_wireless_stick_lite           ESP32    240MHz       4MB      320KB   Heltec Wireless Stick Lite
```
This is in the board list for platformio now, but it has the wrong MCU. I get `A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?` when trying to use the board profile.