platformio/platform-espressif32 #947: ESP32-PICO-V3-02 with 8MB partition table is not booting

slaesh, 2022-10-28:
I am having troubles using the ESP32-PICO-V3-02 with the full flash size of 8MB.
Using the "default.csv" is working fine, but when switching to the "default_8mb.csv" it does not boot any more.. :(

I already had a short chat with @maxgerhardt (thanks again!) [here](https://community.platformio.org/t/support-for-eps32-pico-v3/26967/23?u=slaesh) and he suggested me to file an issue here :)

# repo to reproduce

https://github.com/slaesh/eps32-pico-v3-8mb-flash-issue

# logs using platformio

using platformio and the default partition table, a reset looks like this

```
huhu
huhu
huhu
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   228][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
huhu
huhu
huhu
huhu
```

using platformio and the 8mb partition table, this error loops around:

```
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46
```

# platformio version used

[image]

# artefacts platformio

[platformio_output_8mb.zip](https://github.com/platformio/platform-espressif32/files/9885887/platformio_output_8mb.zip)

[platformio_output_default.zip](https://github.com/platformio/platform-espressif32/files/9885897/platformio_output_default.zip)

# logs using arduino ide

```
Sketch uses 262409 bytes (7%) of program storage space. Maximum is 3342336 bytes.
Global variables use 16520 bytes (5%) of dynamic memory, leaving 311160 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.2.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-PICO-V3-02 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:83:98:d1:a6:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 4.5s
Compressed 18880 bytes to 13028...
Writing at 0x00001000... (100 %)
Wrote 18880 bytes (13028 compressed) at 0x00001000 in 0.2 seconds (effective 648.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 1699.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2616.9 kbit/s)...
Hash of data verified.
Compressed 262800 bytes to 143222...
Writing at 0x00010000... (11 %)
Writing at 0x0001ceb2... (22 %)
Writing at 0x00024e82... (33 %)
Writing at 0x0002a2df... (44 %)
Writing at 0x0002f640... (55 %)
Writing at 0x00037ee7... (66 %)
Writing at 0x00040e33... (77 %)
Writing at 0x00046839... (88 %)
Writing at 0x0004befe... (100 %)
Wrote 262800 bytes (143222 compressed) at 0x00010000 in 2.0 seconds (effective 1057.6 kbit/s)...
Hash of data verified.
```

[image]

serial output arduino ide

```
huhu
huhu
huhu
huhu
huhu
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   228][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
huhu
huhu
huhu
huhu
```

# arduino version used

[image]

# artefacts platformio

[arduino_output.zip](https://github.com/platformio/platform-espressif32/files/9885908/arduino_output.zip)

Jason2866, 2022-10-28:
Try this, your setup with changes in flash size and maximum size:
```
[env]
platform = espressif32
board = esp32dev
framework = arduino
; PSRAM: "Enabled"
; Core Debug Level: "Verbose"
build_flags =
  -DBOARD_HAS_PSRAM
  -DCORE_DEBUG_LEVEL=5
; Partition Scheme: "8M with spiffs"
board_build.partitions = default_8MB.csv
board_build.flash_size  = 8MB
board_upload.maximum_size = 8388608
; upload + monitor sports and speed
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
```

slaesh, 2022-10-28:
thanks @Jason2866 , sadly no changes..

https://github.com/slaesh/eps32-pico-v3-8mb-flash-issue/commit/dca601530b918f9ce30ddd32eb841cc36dfca761

```
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
ets Jul 29 2019 12:21:46
```

slaesh, 2022-10-28:
another test, merge your suggestion with my config

https://github.com/slaesh/eps32-pico-v3-8mb-flash-issue/commit/fdc81f4e23f1edd95eee7da127944a498b38cac7

```
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:46
```

slaesh, 2022-10-28:
omg! thanks @Jason2866 , I just double checked everything.. again!

and I found it.. we had a miss-spelling... oh no :see_no_evil: 

after comparing your suggestion with these files here:

https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32dev.json
https://github.com/platformio/platform-espressif32/blob/develop/boards/denky_d4.json

I saw the differene.. hard to see, but.. it is NOT `board_build.flash_size  = 8MB` .. 

it is `board_upload.flash_size  = 8MB` !!

so we need to do this:

```
[env]
platform = espressif32
board = esp32dev
framework = arduino
; PSRAM: "Enabled"
; Core Debug Level: "Verbose"
build_flags =
  -DBOARD_HAS_PSRAM
  -mfix-esp32-psram-cache-issue
  -DCORE_DEBUG_LEVEL=5
; Flash Mode: "QIO"
board_build.flash_mode = qio
; Partition Scheme: "8M with spiffs"
board_build.partitions = default_8MB.csv

; Flash settings 8MB
board_upload.flash_size  = 8MB
board_upload.maximum_size = 8388608

; upload + monitor sports and speed
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200

```

closed 2022-10-28 by slaesh

slaesh, 2022-10-28:
I am wondering.. do I need `fix-esp32-psram-cache-issue`?

I found this commit.. https://github.com/platformio/platform-espressif32/commit/1dc190d9f77437290a356b9200d19be55b2d6f83

when do we need it, when not @Jason2866 ? =)

Jason2866, 2022-10-28:
@slaesh Only esp32 rev.1 and some rev.2 needs this fix. Since rev.3 it is not needed anymore. Pico variant is a very new version and does not need this fix.