platformio/platform-espressif32 #948: Building ESP32 with Python 3.11 causes ValueError

froqstar, 2022-10-28:
Hi,
when trying to compile [ExpressLRS](expresslrs.org) under Fedora 37/Python3.11, compilation fails with a `ValueError` in platformio code:

```
Building for ESP32 Platform
ValueError: AST node column range (0, 22) for line range (-1, 252) is not valid:
  File "/home/froqstar/.platformio/penv/lib64/python3.11/site-packages/platformio/builder/main.py", line 186:
    env.SConscript(item, exports="env")
  File "/home/froqstar/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/froqstar/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/froqstar/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/build_html.py", line 72:
    build_common(env, "index.html", not '_RX_' in target_name)
  File "/home/froqstar/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/build_html.py", line 57:
    build_html(mainfile, "INDEX_HTML", out, env, isTX)
  File "/home/froqstar/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/build_html.py", line 35:
    template = engine.get_template(mainfile)
  File "/home/froqstar/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/external/wheezy/template/engine.py", line 57:
    self.compile_template(name)
  File "/home/froqstar/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/external/wheezy/template/engine.py", line 110:
    render_template = self.compiler.compile_source(
  File "/home/froqstar/.config/ExpressLRS Configurator/firmwares/github/ExpressLRS/src/python/external/wheezy/template/compiler.py", line 27:
    compiled = compile(node, name, "exec")
```

Building the same target/commit under Fedora 36/Python 3.10 works fine, leading me to believe this is caused by a Python 3.11 incompatibility.
If it helps, a similar exception caused by Python 3.11 was fixed [like this](https://github.com/spyder-ide/spyder-kernels/pull/423/files).

Jason2866, 2022-11-19:
Have you tried to update Platformio core to latest development version?
It should be fixed.

valeros, 2022-12-01:
Closing as the issue is not quite related to the espressif32 platform, it's rather a SCons problem.

closed 2022-12-01 by valeros