platformio/platform-espressif32 #949: Arduino-as-a-component with patch

marchingband, 2022-10-29:
I would like to use esp-idf version 4.4, but with a patch applied which provides eMMC support for ESP32-S3.
The patch is here https://github.com/espressif/esp-idf/issues/8257
Is this possible using PIO and the Arduino lib builder method?
thank you!

Jason2866, 2022-10-29:
Yes, If you use the patched IDF for compiling the Arduino libs.

marchingband, 2022-10-29:
Amazing news thank you!
I have read various threads outlining this process, but it's a little confusing. Is there a step-by-step guide you could recommend for this process of building Arduino-as-a-component under PIO?

marchingband, 2022-10-29:
For example some anecdotal reports imply that a specific commit of idf must be used, to match that of Arduino. Others imply that is not necessarily true any more. Certainly some users report cryptic errors that need to be fixed in inventive ways, it would be nice to have some clarity on the state of affairs as it stands today.

Jason2866, 2022-10-29:
I use IDF branch release/v4.4 without any problems to compile Arduino as component.
You can use the Arduino lib builder to build a fully working Arduino framework.
Just fork idf and the Arduino lib builder, implement your patch and change in the Arduino lib builder the source to use for idf to your fork and start a build.

marchingband, 2022-10-29:
Ok thank you so so much @Jason2866 i appreciate it a lot. That doesn't sound so bad! I will try this out, and if it works maybe I will make a little GitHub gist about it or something. It's a very powerful feature of the PIO ecosystem!

marchingband, 2022-10-31:
It worked so well, thanks again!
Here is a gist with a step-by-step for this specific flow ... patching the esp-idf that Arduino uses.
https://gist.github.com/marchingband/2fc560a5e70c074ba6b0aa0dd130c4b0

valeros, 2022-12-01:
Feel free to reopen if you still any help.

closed 2022-12-01 by valeros