platformio/platform-espressif32 #951: xtensa-esp32s3-elf-g++: not found

cvjensen, 2022-11-02:
I've just bought a ESP32S3 board. It's the red one here:

[image]

But whenever I try to build the firmware in ESPHome I receive the following error:

**sh: 1: xtensa-esp32s3-elf-g++: not found**

The full log is here:
```
INFO Reading configuration /config/esphome/test.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing Hash
INFO Installing Hash
Library Manager: Installing ESP8266WiFi
INFO Installing ESP8266WiFi
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 1.2.7
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32s3-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32s3-elf-g++: not found
sh: 1: xtensa-esp32s3-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32s3-elf-g++: not found
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 5.38 seconds ==========================
```

And my ESPHome config here:
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: 2.0.5
    platform_version: 5.2.0
  variant: esp32s3
```

I have tried with alot of different framework settings but no luck at all.

I'm sorry if this is not the right place to ask/create an issue - If I google the error I only get very very few results, so I'm a little lost.
Bare with me I'm still new to all this ESP stuff
If any information is missing/needed, just ask.

Jason2866, 2022-11-02:
Your board config does not fit to your board `HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash`

cvjensen, 2022-11-02:
> Your board config does not fit to your board `HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash` Which flash and PSRAM type does the board have? QIO / OPI? Which combination?

Where can I see that? All I know is that it is the ESP32-S3 N16R8 from this [link](https://www.aliexpress.com/item/1005004609461716.html?spm=a2g0o.order_list.0.0.21ef1802eIKAmf)

> Start with a simple Platformio Example sketch NOT EspHome!!

Where do I get started ?

Jason2866, 2022-11-02:
If the wroom module is mounted you have marked it is:
8 MB Octal PSRAM, 16 MB Quad SPI Flash.
You have to specify this settings in your env

Jason2866, 2022-11-02:
Probably this will work
```
board_build.partitions = default_16MB.csv
board_build.f_cpu = 240000000L
board_build.arduino.memory_type = qio_opi
board_build.flash_freq = 80m
board_upload.flash_size = 16MB
board_build.flash_mode = dio
board_build.boot = qio
board_build.flash_type = qio
board_build.psram_type = opi
board_build.memory_type = qio_opi
board_build.boot_freq = 80m
```

Jason2866, 2022-11-02:
Afaik it is not easy (impossible?) in Home Assistant to use a other platform version as the supported one. So thats why i wrote use a plain simple Platformio example.
If the example works you have to ask for help in the HA community.

cvjensen, 2022-11-02:
> 

Where should I specify this?

Can I test this using my ESPHome docker container? It has the platformio installed.

I'm just wondering why is fails to build the firmware , 
I'm not trying to upload it directly to the device in ESPHome
I've just tried to build it and download it.

cvjensen, 2022-11-02:
Alright @Jason2866 , thanks for the help so far. 
I will give it a shot with the config you sent and then try to ask the esphome guys.

closed 2022-12-01 by valeros