platformio/platform-espressif32 #952: following dicussion closed --> https://github.com/platformio/platform-espressif32/issues/315

joel14600, 2022-11-02:
Hi all,

I try to use ulp processeur for alarm detector with battery. my goal is to made à 4s delay after gpio_pin level high detected

I saw that "likeablob" has write in issue 315 that he has tested the fork on Linux only.   I have a double boot on my PC and I have install VScode and PlatformIO on my Linux mint.

I test with the sample that I found in the link of likeablob : https://github.com/likeablob/ulptool-pio/tree/feat-pio-integration

at first I had 4 problems.
I have solved 2 by adding initialisation in main.cpp

uint32_t ulp_entry = 0;
uint32_t ulp_count = 0;

it remains 2 problems:
undefined reference to `_binary_ulp_main_bin_end'
undefined reference to `_binary_ulp_main_bin_start'

but there are defined in ulp_main.h

// Unlike the esp-idf always use these binary blob names
extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");

`Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'ulptool-pio/pre_extra_script_ulptool.py'
File "/home/joel/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 186, in <module>
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ulptool-pio @ 0.1.0+sha.6d4c5c5
Building in release mode

warning: Ignoring missing SConscript 'post_extra_script_ulptool.py'
File "/home/joel/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 196, in <module>
Linking .pio/build/esp32dev/firmware.elf
/home/joel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o:(.literal._Z5setupv+0x18): undefined reference to `_binary_ulp_main_bin_end'
/home/joel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o:(.literal._Z5setupv+0x1c): undefined reference to `_binary_ulp_main_bin_start'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
==================================================================== [FAILED] Took 3.54 seconds`

Please if you had the same problems I am interresting on how you have solved the problem.

Best Regards
Joël

Jason2866, 2022-11-02:
Have you seen the [examples](https://github.com/platformio/platform-espressif32/tree/develop/examples) for ULP?
Change your setup (ulp folder has moved when comparing to plain IDF) to the platformio needs. Is CMakeLists.txt adopted?

joel14600, 2022-11-02:
Hi Jason,

Thanks for your answers. No I haven't seen the exemples.  I will see  the exemple /espidf-ulp-pulse/ and try it this night.

valeros, 2022-12-01:
Feel free to reopen if you still any help.

closed 2022-12-01 by valeros