platformio/platform-espressif32 #959: half of the ESP 32 SDK does not work in platformio!

hexffffffff, 2022-11-13:
50% function of the ESP 32 platform don't work! 
The essence of the problem:
espressif delivers its SDK with most of the sources files. But in  platformio ESP32 SDK  for some reason, it is presented in object files, without  .c files. The ESP 32 SDK has a lot of conditional compilation and plug-ins. For example, to use the Internet connection functions via PPP (pppapi_ppos_create, pppapi_set_default pppapi_connect etc (in header "netif/ppp/pppapi.h")), you should do #define PPP_SUPPORT. In the platform for this, you need to write in the platformio.ini file build_flags =-D CONFIG_LWIP_PPP_SUPPORT -D CONFIG_LWIP_PPP_PAP_SUPPORT. Without this, the functions will not be visible in pppapi.h  However, there are no such functions in the platformio object files!!! Therefore, after adding any of these functions, you will get a compilation error  @undefined reference to pppapi_ppos_create , pppapi_set_default etc..."  Naturally, you will receive such an error when connecting any function that is not included in the main package. And this is most likely the majority of the SDK functions!!! 

How I solved the problem:
In my case with PPP I just downloaded the SDK from  espressif  and added a folder from there "netif" to the root of the project. However, I spent a couple of days solving the problem.

It seems to me that this is an unacceptable and gross mistake made by the creators of platformio!

Jason2866, 2022-11-13:
Not related to Platformio, thats how Espressif Arduino is.

hexffffffff, 2022-11-13:
> Not related to Platformio, thats how Espressif Arduino works.. Please do not blame people doing a great job, when you have not reaserched correctly how everything fits together.

I don't quite understand what you're talking about? Are you saying that the creators of platformio did not take the SDK directly from espressif? Are you saying that the creators of platformio took the SDK from the Arduino IDE? And this error is from Arduino IDE?

Jason2866, 2022-11-13:
I say, all stuff is used in original from espressif. There is no file missing.
If you use Arduino framework you cant change settings in sdkconfig, since the libs are ready compiled.

hexffffffff, 2022-11-14:
> I say, all stuff is used in original from espressif. There is no file missing. If you use Arduino framework you cant change settings in sdkconfig, since the libs are ready compiled. If you call this error than yes it is in original Arduino IDE.

I checked if it compiles in Arduino IDE (ver 1.8.19)
Here is the  test code:
#include "netif/ppp/pppapi.h"
struct netif ppp_netif;
 ppp_pcb *ppp;
void setup() {
  ppp = pppapi_pppos_create(&ppp_netif, NULL, NULL, NULL);
   pppapi_set_default(ppp);
}
void loop() {
}

Everything compiles wonderfully. Therefore, your error statement in the Arduino IDE is not true.

valeros, 2022-12-01:
Hi @hexffffffff , what version of the EPS32 core are you using? Please provide a minimal project to reproduce the issue.

hexffffffff, 2022-12-01:
> Hi @hexffffffff , what version of the EPS32 core are you using? Please provide a minimal project to reproduce the issue.

Espressif 32 -5.2.0
core v4.4.2
minimal project (I already wrote in the last post): 
//=====================================
#include "netif/ppp/pppapi.h"
struct netif ppp_netif;
ppp_pcb *ppp;
void setup() {
ppp = pppapi_pppos_create(&ppp_netif, NULL, NULL, NULL);
pppapi_set_default(ppp);
}
void loop() {
}
//=====================================
platformio.ini source:
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
build_flags = 
    -D CONFIG_PPP_SUPPORT
    -D CONFIG_LWIP_PPP_SUPPORT

valeros, 2022-12-01:
Can you upload the verbose build log from Arduino IDE to https://pastebin.com ?

valeros, 2022-12-01:
No wonder it doesn't work. You're using an outdated Arduino core for ESP32 v1.0.6, while the latest one is 2.0.5. The last known version of the `espressif32` platform that supports the Arduino core v1.0.6 is 3.5.0. If you don't want to update your project to support the latest Arduino core then just lock your platform version to 3.5.0:
```
[env:esp32doit-devkit-v1]
platform = espressif32 @ ~3.5.0
board = esp32doit-devkit-v1
framework = arduino
build_flags =
    -D CONFIG_PPP_SUPPORT
    -D CONFIG_LWIP_PPP_SUPPORT
```

closed 2022-12-01 by valeros

hexffffffff, 2022-12-01:
Wait! I have sent you a log from  Arduino IDE (as you requested). And everything works in the Arduino IDE. But the problem is observed in platformio. I used version 5.2.0

valeros, 2022-12-02:
You're confusing the platform version (5.2.0) and the framework version (2.0.5). In your Arduino setup you're using the framework version 1.0.6 which is almost 2 years old. If you update your framework to the latest version in Arduino IDE, you will face the same issue with undefined symbols. If you want to use the framework version 1.0.6 then you need to downgrade your platform installation to v3.5.0 as I described above:

```ini
[env:esp32doit-devkit-v1]
platform = espressif32 @ ~3.5.0
board = esp32doit-devkit-v1
framework = arduino
build_flags =
    -D CONFIG_PPP_SUPPORT
    -D CONFIG_LWIP_PPP_SUPPORT
```

hexffffffff, 2022-12-02:
> No wonder it doesn't work. You're using an outdated Arduino core for ESP32 v1.0.6, while the latest one is 2.0.5. The last known version of the `espressif32` platform that supports the Arduino core [...]

I know that in version to 3.5.0 it worked in platformio . The problem appeared in the latest versions of platformio.  version 3.5.0 not require flags  CONFIG_PPP_SUPPORT CONFIG_LWIP_PPP_SUPPORT and compiled this code correctly. Perhaps the problem appeared in the latest versions of the core. The problem is related to conditional compilation in  espidf. But this does not mean that the problem does not need to be fixed. Arduino ide is positioned as a toy for children. But platformio it is positioned as a professional solution and it is unacceptable that half of the sdk do not work there.

medvosa, 2022-12-19:
Hi @hexffffffff, you should furst rename your .c Giles to .s
Second - platformio is intended to write code, not to compile it. Please use Arduino idé for building process

Also DO not use conditional compilation, build everythibg instead