platformio/platform-espressif32 #967: Could not initialize project ESP32-S3

SynonZelra, 2022-11-25:


ggalisky, 2022-11-29:
I also am having this issue, working on fixing it

ivankravets, 2022-11-29:
@valeros, could you help here?

martinberlin, 2022-11-29:
In case it helps anyone I could make it work with S3 correctly.
There is a working example including platformio.ini and custom board in my [ESP32S3 Epub reader fork](https://github.com/martinberlin/IT8951-esp32-epub-reader).

valeros, 2022-12-02:
Hi @martinberlin, I guess your restriction on the platform version `platformio/espressif32@^4.3` is the reason why the board is not found. ESP-IDF support for S3 was added in the platform version v5.0. As for the docs, it implies that the latest development platform is used. 

Anyone who is facing the similar issues, please double check that you're using the latest version of the `espressif32` platform.

closed 2022-12-02 by valeros