platformio/platform-espressif32 #968: pio pkg exec -- esptool.py broken in latest version

j4m3s, 2022-11-26:
using tool-esptoolpy@1.40201.0, the `pio pkg exec -- esptool.py -h` command (or any other esptool command) produces the following error output:
```
Using tool-esptoolpy@1.40201.0 package
Error: [Errno 13] Permission denied: 'esptool.py'
```

using tool-esptoolpy@1.30300.0 it works fine.

The issue appears to be that the `tool-esptoolpy/esptool.py` file is not executable in this version where it was before.  Comparing other files it looks like `espefuse.py`, `espsecure.py` and `esp_rfc2217_server.py` have also lost their executable bits.

My apologies if I'm filing this bug in the wrong repo, I couldn't see separate repos for the tools, only the platforms.

ivankravets, 2022-11-29:
Thanks for the issue. Resolved in https://github.com/platformio/platform-espressif32/commit/c0df90fda865f592b298bad95a02cb59f75644dc

closed 2022-11-29 by ivankravets

j4m3s, 2023-04-13:
Hi @ivankravets 

I've come to use `espefuse.py` and it still has the problem, the exec bit isn't set.  Same for the other scripts in the same directory.  I can't see how to reopen the issue - do you want me to open a new one?

ivankravets, 2023-04-14:
Do you use the latest version of the package? Could you try the following command?
```
pio pkg exec --package "platformio/tool-esptoolpy@>=1.40501.0" -- espefuse.py -h
```

j4m3s, 2023-04-16:
Running that version works fine.  I have to admit in between submitting my question and your reply I manually changed the permissions on the files and now, however I run espefuse I seem to get v4.5.1 so I'm no use to you for testing either!

It's quite possible that I was using an old version - please ignore me, I'll open a new issue if there's a genuine problem when I next do a fresh install.