platformio/platform-espressif32 #974: ESP32-S3 Filesystem Upload Issue

ams-hh, 2022-12-03:
Configuration: Windows 10, Visual Studio Code 1.71.2, Platform ESP32 5.2.0

Hello!
I started using a ESP32-S3 (as ESP-S3-WROOM-1 4 and 16MB  = ESP32-S3-WROOM-1N16)

In opposite to ESP32(-WROOM-32) with 4MB I have never managed to upload the filesystem image. Means: All the file in the \data directory bundled as "littlefs.bin", as done by the platformio system when clicking "Upload Filesystem Image".

With ESP32 everything works well. With ESP32-S3 long breaks within the stream occur and after some time the upload is canceled.

``
Uploading G:\ESP-Server\.pio\build\esp32S3-COM3\littlefs.bin
esptool.py v4.2.1
Serial port COM3
Connecting.........
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:53:6c:0c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00290000 to 0x003fffff...
Compressed 1507328 bytes to 52514...
Writing at 0x00290000... (25 %)
Writing at 0x002fb61b... (50 %)
Writing at 0x003120aa... (75 %)

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
*** [uploadfs] Error 2

`
(Yes, I know what "Possible serial noise" means. But on the same Hardware (selfmade development board) all the ESP8266-12 and ESP32 work fine while uploading.)

When it the comes to an 16MB ESP32-S3 with e.g. the predefined partition csv "large_spiffs_16MB.csv", the littlefs.bin file is abt. 7MB large. Impossible to upload it. I've tried it as well from "outside" of platformio with 
esptool.py --port COM4 write_flash 0x910000 littlefs.bin. (Address from "large_spiffs_16MB.csv", spiffs-Partition) Same result. Upload canceled after abt. 33-55%. (I tried reducing the Baudrate with --baud. No change)

Does someone has an idea what to try?

Thank you in advance
Andree

Jason2866, 2022-12-06:
@ams-hh Do you have an example configuration somewhere hosted?

ams-hh, 2022-12-06:
Hello Jason2866,

not quite sure what is meant with "example configuration". 
If its platformio.ini and boards-json, I insert it in the following:

========== platformio.ini: ==========
`
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = esp32S3-16MB
lib_dir = g:\projekte\PIO\Libraries\Common
boards_dir = g:\projekte\PIO\Boards

[env]
framework = arduino
monitor_filters = time, default

[env:esp32S3-16MB]
build_flags = 
  -D ESPFAM_32
  -D ESPMOD_S3W1 
  -D FS_LITTLEFS
  -Ig:/projekte/pio
  -Ig:/projekte/pio/libraries
  -Ig:/projekte/pio/MScadaWS/include/ws
  -Ig:/projekte/pio/MScadaWS/include/appli
  -D CORE_DEBUG_LEVEL=0
platform = espressif32
board = esp32s3_16MB_PS0_FS6F0000
lib_extra_dirs = 
  g:\projekte\PIO\Libraries\ESP32
board_build.filesystem = littlefs
`

========== esp32s3_16MB_PS0_FS6F0000.json: ==========
`

{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "partitions": "large_spiffs_16MB.csv",
      "memory_type": "qio_qspi"
    },
    "core": "esp32",
    "extra_flags": 
    "-DARDUINO_USB_MODE=1",
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "ESP32-S3 16MB No PSRAM",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://blabla",
  "vendor": "espressif"
}
`

With this, when clicking on "Upload File System Image" a 7MB (or so) File littlefs.bin
should be created in .pio\build\esp32S3-16MB and SHOULD be loaded into the ESP.
And as mentioned, I thought the same can be tried with
esptool.py --port COMx write_flash 0x910000 littlefs.bin. 

Thank you!!
A.

Jason2866, 2023-01-12:
Maybe using latest development esptool.py helps. There was a bug in esptool.py stub loader for the S3 handling 16MB flash. Still faulty since 32MB can still not be used.
It is not a Platformio Bug!

ams-hh, 2023-01-16:
Hello Jason2866,
thank you! I will give it a try.
Regards
A.

BlueAndi, 2023-03-16:
See #1051 for the current solution.

closed 2023-05-08 by stale[bot]