platformio/platform-espressif32 #978: Add board WT32-SCO and WT32-SCO-Plus

zackees, 2022-12-14:
Does anyone know which board I can select as a work around?

I can compile and flash but `esp_highmem_get_phys_size()` and `esp_himem_get_free_size()` both crash when called.

```
  esp_himem_get_phys_size();
  esp_himem_get_free_size();
```

Results in

```
E (153) spiram: SPI RAM not initialized

abort() was called at PC 0x400df714 on core 0

Backtrace:0x40083671:0x3ffe3ae00x400892a1:0x3ffe3b00 0x4008df6d:0x3ffe3b20 0x400df714:0x3ffe3ba0 0x400df753:0x3ffe3bc0 0x400df63f:0x3ffe3be0 0x400db8f7:0x3ffe3c10 0x40082d8d:0x3ffe3c40 0x400792ba:0x3ffe3c90  |<-CORRUPTED

  #0  0x40083671:0x3ffe3ae0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x400892a1:0x3ffe3b00 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x4008df6d:0x3ffe3b20 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x400df714:0x3ffe3ba0 in esp_spiram_get_chip_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/spiram.c:145      
      (inlined by) esp_spiram_get_chip_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/spiram.c:141
  #4  0x400df753:0x3ffe3bc0 in esp_spiram_get_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/spiram.c:240
  #5  0x400df63f:0x3ffe3be0 in esp_himem_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/esp_himem.c:134
  #6  0x400db8f7:0x3ffe3c10 in do_global_ctors at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:185
      (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:404
  #7  0x40082d8d:0x3ffe3c40 in call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:648
  #8  0x400792ba:0x3ffe3c90 in ?? ??:0
```

zackees, 2022-12-14:
This happens for the following `platformio.ini` configurations:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:WT32-SC01]
platform = https://github.com/platformio/platform-espressif32.git
board = wt32-eth01  # Note that to use psram board must use esp32dev
framework = arduino
upload_speed = 921600
monitor_speed = 115200
monitor_filters = 
	default
	esp32_exception_decoder
;board_build.f_flash = 80000000L
build_flags =
  -DCORE_DEBUG_LEVEL=2
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-I src/
;build_flags =
;  -DUSER_SETUP_LOADED=1
;  -DST7796_DRIVER=1
;  -DTFT_WIDTH=480
;  -DTFT_HEIGHT=320
;  -DUSE_HSPI_PORT=1
;  -DPIN_SDA=18
;  -DPIN_SCL=19
;  -DTFT_MISO=12
;  -DTFT_MOSI=13
;  -DTFT_SCLK=14
;  -DTFT_CS=15
;  -DTFT_DC=21
;  -DTFT_RST=22
;  -DTFT_BL=23
;  -DLOAD_GLCD
;  -DLOAD_FONT2
;  -DLOAD_FONT4
;  -DLOAD_FONT6
;  -DLOAD_FONT7
;  -DLOAD_FONT8
;  -DLOAD_GFXFF
;  -DSPI_FREQUENCY=27000000
;  -DSPI_READ_FREQUENCY=20000000
;  -DSPI_TOUCH_FREQUENCY=2500000
;  -DSMOOTH_FONT
;lib_deps =
;    Bodmer/TFT_eSPI
```

And also:

```
board = esp-wrover-kit
```

And also:

```
board = esp32dev
```

I'm using VSCode with the arduino framework. It happens with the release:

```
platform = https://github.com/platformio/platform-espressif32.git
```

And also:

```
platform = espressif32
```

The error says that it can't enable spiram, which is interesting because the WT32-SCO uses psram. I can access and use the first 4mb of psram on the chip, which I've confirmed works with a memory test:

```
 void *memory = ps_malloc(100000);
 memset(memory, 0, 100000);
 free(memory);
```

This issue only affects the upper 4mb high memory.

Jason2866, 2022-12-14:
You cant address more than 4MB PSRAM with the ESP32.

zackees, 2022-12-14:
The internet reports that people are accessing the upper 4mb psram through high mem bank switching. At the very least calling this function should not crash the system.

Jason2866, 2022-12-14:
Anyways, not a platformio issue.

zackees, 2022-12-14:
Does anyone know if this is issue is fixed with ESP-IDF v5.X?

I see that platformio latest (5.2.0) currently uses ESP-IDF v4.4.2 and proper support for the WT32-SCO board might have come after ESP-IDF v.4.4.2. It could just be a fix waiting for an upstream merge with the new toolchain.

sukesh-ak, 2022-12-15:
@zackees 
You can use this project for WT32-* boards. Once PIO support for IDF 5.0 happens, will add the support in the project.
https://github.com/sukesh-ak/ESP32-TUX/

sukesh-ak, 2022-12-15:
@zackees 
And ESP-IDF himem sample is here
https://github.com/espressif/esp-idf/tree/release/v5.0/examples/system/himem

ThomasIJsseldijk, 2023-11-07:
hi, i was wondering if there is any progress adding these boards?