platformio/platform-espressif32 #979: Feature Request: Support for IDF 5.0

Jason2866, 2022-12-15:
Espressif released IDF 5.0 a while ago.

zapta, 2022-12-28:
Not sure how to vote for it so supporting it here with a comment.

IDF 5 has support for continuous ADC adc_continuous.h which we need.  

Having to develop with an old SDK that doesn't match the latest documentation and feature set set is difficult.

Pespiri, 2022-12-29:
> Having to develop with an old SDK that doesn't match the latest documentation and feature set set is difficult.

The documentation for ESP-IDF 4.4.2 can be found here ([ESP-IDF 4.4.2 Docs](https://docs.espressif.com/projects/esp-idf/en/v4.4.2))  
You can change target and version on the top right

valeros, 2022-12-29:
Integration of IDF v5.0 is already in the works and will be publically available next week.

grisharav, 2022-12-31:
Thanks, shall this release include toolchain upgrade to GCC 11.2?

Jason2866, 2022-12-31:
IDF 5.0 won't work without actual tool chains, it is a dependency. So, yes.

closed 2023-01-06 by valeros

zapta, 2023-01-06:
Can you please add more information on how to use it?  Looking at platformio's documentation it still mentions V4 as the current version.

https://docs.platformio.org/en/latest/frameworks/espidf.html

grisharav, 2023-01-08:
I also don't see a release, is it upcoming?

valeros, 2023-01-09:
If nothing goes wrong, a new release will be published by the end of this week. The latest changes with IDF 5.0 can be pulled via the upstream version of the development platform, e.g.:
```ini
[env:esp32dev]
board = https://github.com/platformio/platform-espressif32.git
platform = espressif32
framework = espidf
```
Note: You should have a Git client installed in your system.

flavioroth, 2023-01-16:
The snippet from @valeros doesn't work me. It breaks with the error: `Error: Unknown board ID 'https://github.com/platformio/platform-espressif32.git'` 

Did you mean ?
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = espidf
```

Coobik, 2023-01-17:
hi all! i have updated Espressif 32 platform to version 6.0.0 in platformio in vscode
build fails with error:
```
 File "/Users/USER/.platformio/packages/framework-espidf/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "/Users/USER/.platformio/packages/framework-espidf/tools/ldgen/ldgen/entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/Users/USER/.platformio/penv/lib/python3.9/site-packages/pyparsing.py)
Compiling .pio/build/esp32dev/bootloader/bootloader_support/src/flash_partitions.o
*** [.pio/build/esp32dev/sections.ld] Error 1
```
--
i have esp-idf toolchain (v5) configured and it is able to build the project from command line
any suggestions?
thanks!

valeros, 2023-01-18:
Hi @Coobik, please open a new issue with detailed instructions on how to reproduce the issue.

ghost711, 2023-01-23:
@Coobik , I just got past the same issue.  Just had to activate the platformio virtualenv, and then update pyparsing. 
On my system that looks like this:
```
source ~/.platformio/penv/bin/activate
pip install -U pyparsing
```

MrSniffer, 2023-01-23:
When will I be able to access ESP32 5.0 APIs using platform.io and framework=arduino? Right now I am stuck at ver 4.x APIs.

Jason2866, 2023-01-23:
@MrSniffer  First espressif Arduino crew needs to port Arduino based on IDF 5.0. The started to work on this.

MrSniffer, 2023-03-20:
Is the espressif Arduino work ready for prime time, so we can use IDF 5.0 with Arduino framework projects under Platformio?

Jason2866, 2023-03-20:
The started the work. Probably late this year

commit ecc617e3 (2023-01-06): Add support for IDF v5.0

Resolves #979