platformio/platform-espressif32 #984: platform-esressif32 v5.3.0 brokes BluetoothSerial

Jaimi5, 2022-12-29:
After updating to 5.3.0 the BluetoothSerial register_callback is not working as before. It generates the following error:

```error: no matching function for call to 'BluetoothSerial::register_callback(void (&)(esp_spp_cb_event_t, esp_spp_cb_param_t*))'```

I do not know if it is an actual error or there is something that has been updated. 
Using the previous version 5.2.0 it works correctly as before.

Hope this help!
Thanks a lot for your support!

valeros, 2022-12-29:
Hi @Jaimi5 ! Could you please try the same code in the Arduino IDE with the latest ESP32 core? Does it work there?

Jason2866, 2023-01-12:
Fix PR has been merged https://github.com/espressif/arduino-esp32/pull/7681

closed 2023-01-12 by Jaimi5

seconika, 2023-03-23:
Hi,
Any news regarding this?   The "fix" do not work for me..