platformio/platform-espressif32 #985: Flashing a UM TinyS3 with 5.3.0 fails

colde, 2022-12-30:
After uprading to v5.3.0 i can no longer flash an attached TinyS3.

Here is the log of it failing:
```
Executing task in folder RotatingLED-TinyS3: platformio run --target upload 

Processing um_tinys3 (platform: espressif32; board: um_tinys3; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/um_tinys3.html
PLATFORM: Espressif 32 (5.3.0) > Unexpected Maker TinyS3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FastLED @ 3.5.0
|   |-- SPI @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|   |-- WiFi @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/um_tinys3/firmware.elf
Checking size .pio/build/um_tinys3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.9% (used 19232 bytes from 327680 bytes)
Flash: [=         ]   7.2% (used 240885 bytes from 3342336 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbmodem14301
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem14301
Waiting for the new upload port...
Uploading .pio/build/um_tinys3/firmware.bin
esptool.py v4.4
Serial port /dev/cu.usbmodem14301
```

It seem to get stuck there for however long i wait until i manually terminate it.

With v5.2.0 it works fine, here is the log for that:
Executing task in folder RotatingLED-TinyS3: platformio run --target upload 

Processing um_tinys3 (platform: espressif32@5.2.0; board: um_tinys3; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/um_tinys3.html
PLATFORM: Espressif 32 (5.2.0) > Unexpected Maker TinyS3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
[... 138 lines cut ...]
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 473.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 632.1 kbit/s)...
Hash of data verified.
Compressed 224272 bytes to 123649...
Writing at 0x00010000... (12 %)
Writing at 0x0001c9e9... (25 %)
Writing at 0x00022b7b... (37 %)
Writing at 0x0002809d... (50 %)
Writing at 0x0002d526... (62 %)
Writing at 0x00034e67... (75 %)
Writing at 0x0003dee6... (87 %)
Writing at 0x0004381f... (100 %)
Wrote 224272 bytes (123649 compressed) at 0x00010000 in 2.3 seconds (effective 781.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================================ [SUCCESS] Took 27.88 seconds ============================================================
 *  Terminal will be reused by tasks, press any key to close it.

Jason2866, 2022-12-30:
Flashing worked perfectly fine regarding your log file

Jason2866, 2022-12-31:
Maybe this https://github.com/espressif/arduino-esp32/pull/7631

colde, 2023-01-02:
I'm not sure why it would be that issue, as i don't get a warning like that, and they have the issue with 5.2.0 which works for me.

Jason2866, 2023-01-02:
So, the only relevant change I see is the change of the esptool.py version.
You can try the version used before.
In general flashing the S3 via the inbuilt USB modem does work, with the latest release.

colde, 2023-01-02:
Yeah, i suspected it was likely related to the esptool.py version. Is there a way i can configure platformio, to use the new framework, but the older esptool.py, or do you suggest trying to just build a binary and manually flash it?

valeros, 2023-01-06:
Hi @colde ! Does it work in the Arduino IDE? Anyway, you can find available versions of the esptool package [here](https://registry.platformio.org/tools/platformio/tool-esptoolpy/versions) and set the desired version directly in your `platformio.ini`:
```ini
[env:your_env]
platform = espressif32
...
platform_packages = 
    platformio/tool-esptoolpy @ 1.40300.0
```

colde, 2023-01-07:
Strange, for whatever reason it seems to work now, after having flashed it with espressif32@5.3.0 and the esptool 1.40300.0, meaning i can also flash it with 1.40400.0 now. I guess i'll close this as i can no longer reproduce it.

closed 2023-01-07 by colde