platformio/platform-espressif32 #986: esp_tls_conn_write fails after upgrading to platformio 5.3.0 and espidf 4.4.3

gmdriscoll, 2022-12-30:
Framework selection in platformio.ini espressif32@4.4.0 has been working for many months with appropriate changes using ESP_NETIF_WIFI_DNS_MAIN, etc.

Problem is as below when upgraded to platformio 5.3.0 and espidf 4.4.3:
Using HTTPS with server_root_cert_pem_start for POST.

      esp_tls_cfg_t cfg = {};
      cfg.cacert_buf = (const unsigned char *) server_root_cert_pem_start;
      cfg.cacert_bytes = server_root_cert_pem_end - server_root_cert_pem_start;

HTTPS connection established properly with no errors.
Can change the host in the POST request to be "www.google.com" and no connection established and correct errors returned.

Initially making https connection with:
   combURL = "https://www.xxxxxx-xxx.com/path"   //NOTICE THE "-" in the URL
   (also tried with combURL = "www.xxxxxx-xxx.com" )
   struct esp_tls *fb_tls = esp_tls_conn_http_new(combURL, &cfg);
   if (fb_tls != NULL){  ... }

then using this post string:
    POST /data.asmx/Device HTTP/1.1
    Host: www.xxxxxx-xxx.com:443
    Content-type: application/json
    Content-length: 96
    
    {JSON PACKET HERE}

Then writing the post string:
 do{
    ret = esp_tls_conn_write(fb_tls,
                            std_request.c_str() + written_bytes,
                            std_request.size() - written_bytes);
while (written_bytes < std_request.size());

As soon as it hits the esp_tls_conn_write command, the code panics with:
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0xc83e29f9  PS      : 0x00060f30  A0      : 0x800dc9e0  A1      : 0x3ffe6770
A2      : 0xc83e29f9  A3      : 0x3f4047f4  A4      : 0x00000000  A5      : 0x3ffe6780
A6      : 0x3ffe6780  A7      : 0x3ffe6b00  A8      : 0x800e7f4c  A9      : 0x3ffe6720
A10     : 0x3ffe2aa0  A11     : 0x3ffbe74c  A12     : 0x000000ef  A13     : 0x000077b0
A14     : 0x3f4047f4  A15     : 0x3ffbe74c  SAR     : 0x00000004  EXCCAUSE: 0x00000014
EXCVADDR: 0xc83e29f8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

Backtrace: 0x483e29f6:0x3ffe6770 0x400dc9dd:0x3ffe7520 0x400e61ef:0x3ffe77e0 0x4008d525:0x3ffe7920

  #0  0x483e29f6:0x3ffe6770 in ?? ??:0

I have back rev'ed to 4.4.0 again as this has been working. 

Any help or ideas would be greatly appreciated.  Wondering if it is a parsing bug in esp_tls_conn_write with the "-" in the URL?

Thanks,
-Greg

Jason2866, 2022-12-30:
Probablynot a platformio issue, the IDF has many changes, so there is no guarantee that older code is still working with the new IDF version.

valeros, 2023-01-06:
Hi @gmdriscoll ! Does it work in the Arduino IDE?

closed 2023-02-19 by stale[bot]