platformio/platform-espressif32 #987: Linker problem with typeinfo for fs::FileImpl in version 5.3.0 & arduino core 2.0.6

rvbc1, 2022-12-30:
All files are compiling ok, but linker returns error as below
`.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: build/local/esp-idf/framework-arduinoespressif32/libframework-arduinoespressif32.a(vfs_api.o):(.rodata._ZTI11VFSFileImpl[typeinfo for VFSFileImpl]+0x8): undefined reference to `typeinfo for fs::FileImpl'`
On Version 5.2.0 & Arduino core 2.0.5 code is compiling ok

Jason2866, 2022-12-30:
Example Sketch? There are code changes in Arduino for this part. You probably have to adopt your code.

alex-code, 2023-01-04:
I just updated to 5.3 and get this error too on code that previously compiled ok.

Googling for solutions I found mention of RTTI.
I'm using `build_unflags = -fno-rtti` as there's use of `dynamic_cast`

Possibly related https://github.com/espressif/arduino-esp32/issues/7585

alex-code, 2023-01-05:
> @alex-code The toolchains (compiler) does not support `rtti`. You cant do `build_unflags = -fno-rtti`

Been working for me before 5.3

This compiles on 5.2
```ini
; Condensed
platform = espressif32@5.2.0
build_flags =
	-std=gnu++17
build_unflags =
	-std=gnu++11
	-fno-rtti
```
This fails on 5.3
```ini
; Condensed
platform = espressif32@5.3.0
build_flags =
	-std=gnu++17
build_unflags =
	-std=gnu++11
	-fno-rtti
```

`c:/users/me/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/firebeetle2/lib0e2/libFS.a(vfs_api.cpp.o):(.rodata._ZTI11VFSFileImpl[_ZTI11VFSFileImpl]+0x8): undefined reference to typeinfo for fs::FileImpl' `

Jason2866, 2023-01-05:
Exactly, toolchains changed from 5.2.0 to 5.3.0 from patch3 to patch5.
rtti was never supported correctly in the toolchains. Espressif fixed this and removed the faulty rtti support in patch5 tool chains.
Anyways this is not a platformio issue, since Arduino IDE behaviour is the same.

valeros, 2023-01-06:
Thanks @Jason2866 for clarifying. Indeed seems like the issue is caused by updated toolchains which is out of our control. Anyone who's experiencing the same issue please update your project configuration as described above.

closed 2023-01-06 by valeros

LK-Simon, 2024-02-08:
This is **not** resolved, and is indeed a serious problem!

Just because your library doesn't use RTTI, doesn't mean there aren't plenty of us with code that DOES reliably and with necessity use RTTI.

Jason2866, 2024-02-08:
@LK-Simon At the end the issue was something different ;-).
The fix https://github.com/espressif/arduino-esp32/pull/8768

Currently not in any released core. Fix will be included in core 3.0.0 and the fix is backported in bug fix core 2.0.15. Whenever both core will be released. As long this is not happened you have to do the change locally.

LK-Simon, 2024-02-09:
> @LK-Simon At the end the issue was something different ;-). The fix [espressif/arduino-esp32#8768](https://github.com/espressif/arduino-esp32/pull/8768)
> 
> Currently not in any released core. Fix will be included in core 3.0.0 and the fix is backported in bug fix core 2.0.15. Whenever both core will be released. As long this is not happened you have to do the change locally.

Hi @Jason2866, how do I get Platformio to use the fixed version of the core? :) This has been driving me nuts because my Event Engine absolutely requires RTTI, and I have other issues that make using an older version of the ESP lib untenable.

Jason2866, 2024-02-09:
@LK-Simon This should work:
```
platform = espressif32 @ 6.5.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#release/v2.x
```

LK-Simon, 2024-02-10:
> @LK-Simon This should work:
> 
> ```
> platform = espressif32 @ 6.5.0
> platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#release/v2.x
> ```

You, sir, are a life-saver!

mathemaphysics, 2024-07-06:
This is madness.

Who in their wild insanity disables RTTI. I'm sick of this hellhole.

mathemaphysics, 2024-07-07:
This is an error on their part. I've overridden the build options to use -frtti everywhere. It works. Their compiler does not find this single typeinfo. This is the only one. Just `libFS.a` can't find `fs::FileImpl`'s typeinfo. That's it. Nothing else.

This looks to be a compiler issue.

Jason2866, 2024-07-07:
@mathemaphysics you have to add this change in your framework https://github.com/espressif/arduino-esp32/pull/8768/files