platformio/platform-espressif32 #992: SPIFFS broken with v5.3.0: error -10025

Paolo97Gll, 2023-01-06:
I own an ESP32-WROOM-32 and with the platform version v5.3.0 the SPIFFS is broken. I use the arduino framework, my code has been the same for months and I never had any problems. After upgrading from v5.2.0 to v5.3.0, the command `SPIFFS.begin()` from the library `SPIFFS.h` fail by returning `false` and an error code `-10025`.  Downgrading to v5.2.0 solved.

valeros, 2023-01-09:
Hi @Paolo97Gll ! What's the precise board ID you're using? The main change in the v5.3.0 is updated Arduino core v2.0.6. It'd be great if you could try the same example but in the Arduino IDE.

Paolo97Gll, 2023-01-09:
The board is a [PRODINo ESP32 v1](https://kmpelectronics.eu/products/prodino-esp32-v1/), and I have `board = esp32dev` in my `platformio.ini` file. I will try in the next weeks with the Arduino IDE, I do not have the board with me since I use it in a remote environment.

mkalkbrenner, 2023-01-14:
I just faced something similar. I just ported a project from Arduino IDE to platformio. platform-espressif32 is version v5.3.0.
After the first upload of the data folder, SPIFFS seems to be corrupted.
But even if I upload the old version of the project using the Arduino IDE again, I can't get it to work anymore. So far I don't know how to repair SPIFFS.
```
framework = arduino
platform = espressif32
board = esp32dev
```

mkalkbrenner, 2023-01-14:
OK, I can confirm that the downgrade to 5.2.0 "solves" the issue:
```
framework = arduino
platform = espressif32@5.2
board = esp32dev
```

mkalkbrenner, 2023-01-16:
I don't know what exactly solved the issue, but we now migrated to LittleFS and use qio as flash method. And we replaced the default partiton table by an optimized one.
Now it also works with ^5.3.0.

Jason2866, 2023-02-15:
The underlying issue was a not correct builded or flashed bootloader.
What you have encountered was flash writing errors, caused by not correct set flash mode.
Using latest framework and a full erase of the used device and reflashing will solve.
The issue needed some time to fix. Bugs in Arduino and platformio code. All solved now.

Jason2866, 2023-02-15:
@mkalkbrenner Not an good idea to build a combined firmware https://github.com/PPUC/ZeDMD_ESP32/blob/main/.github/workflows/ZeDMD.yml
By doing this you have to take care that the device is flashed 100% with the same command as it was build. Using a slightly different command set it will behave erratic and your flash problems are back!

closed 2023-03-20 by Paolo97Gll