platformio/platform-espressif32 #1000: Error using "New project" for ESP32-CAM AI Thinker board with ESP-IDF framework

AgainPsychoX, 2023-01-15:


valeros, 2023-01-16:
Hi @AgainPsychoX ! I don't see any obvious errors in the logs, so let's try debugging it the CLI way. Please open your project in the IDE, then open the terminal via `` ctrl+ shift + ` ``, then clean `pio run -t clean` and build `pio run` your project.

AgainPsychoX, 2023-01-16:
My issue isn't about code failing to build, its about errors when creating new project.

As far as I remember, it compiles & upload fine for me. It was lacking VS Code generated files (`c_cpp_properties.json`), but I managed to recreate them by using `pio init --ide=vscode`.

maxgerhardt, 2023-01-16:
Possibly related: https://community.platformio.org/t/creating-new-esp-idf-project-fails-with-modulenotfounderror-no-module-named-idf-component-manager-message/31511?u=maxgerhardt

maxgerhardt, 2023-01-17:
Even PlatformIO-core CI fails with the same error in the this run: https://github.com/platformio/platformio-core/actions/runs/3913391485/jobs/6689228904

> 'ERROR: Could not install packages due to an OSError: [WinError 5] Access is denied: \'D:\\\\a\\\\platformio-core\\\\platformio-core\\\\.tox\\\\testexamples\\\\Lib\\\\site-packages\\\\~harset_normalizer\\\\md.cp39-win_amd64.pyd\'\r\nCheck the permissions.\r\n\r\
>ModuleNotFoundError: No module\r\n  named \'idf_component_manager\')\

What is is up with that `~harset_normalizer` package?

maxgerhardt, 2023-01-17:
Possibly related too: https://github.com/Ousret/charset_normalizer/issues/253

Jason2866, 2023-01-17:
Have you tried latest release 6.0?
There are changes, which could solve this issue.
The issue occurred here too https://github.com/platformio/platform-espressif32/actions/runs/3881735765/jobs/6684846018
In later CIs the error is gone

maxgerhardt, 2023-01-17:
Even if IDFv5 integration in `espressif32@6.0.0` is working, that would mean ESP-IDF in `espressif32@5.3.0` does not work.. ever.

Jason2866, 2023-01-17:
@maxgerhardt If the reason is known a backport can be down.

Jason2866, 2023-01-24:
This backport https://github.com/tasmota/platform-espressif32/commit/ded00a7652962ec17350f713d37f71a22a7274f3 from release 6.0.0 seems to fix the issue.

closed 2023-03-25 by stale[bot]