platformio/platform-espressif32 #1002: Firebeetle DFR0654

imabot2, 2023-01-15:


valeros, 2023-01-16:
Hi @imabot2 ! It seems this board is not supported by [the official Arduino core for ESP32](https://github.com/espressif/arduino-esp32). If you want you can submit a PR with proper pinout configuration to that repository so that it will be available in PlatformIO as well.

Torxgewinde, 2023-01-18:
With Arduino-IDE and the Arduino core for ESP32 the Board selecting the Board "Firebeetle ESP32" compiles working firmware, a compatible/matching board is defined in https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L7063

Here is a picture of the settings to compile for DFR0654: [image]

I am not sure about the proper pin definitions though, the definition might initially just be for the [Firebeetle DFR0478](https://wiki.dfrobot.com/FireBeetle_ESP32_IOT_Microcontroller(V3.0)__Supports_Wi-Fi_&_Bluetooth__SKU__DFR0478)

imabot2, 2023-01-19:
> Hi @imabot2 ! It seems this board is not supported by [the official Arduino core for ESP32](https://github.com/espressif/arduino-esp32). If you want you can submit a PR with proper pinout configuration to that repository so that it will be available in PlatformIO as well.

What is a PR and how to do that ? Any example somewhere ?

valeros, 2023-01-20:
@imabot2 See the [Contributions Guide](https://docs.espressif.com/projects/arduino-esp32/en/latest/contributing.html) and [an example](https://github.com/espressif/arduino-esp32/pull/7523).

ejagombar, 2023-01-28:
Hi, I have also looking into this issue.

The board for the configuration "dfrobot_firebeetle2_esp32s3" does not seem to exist. The .json file links to the webpage for the firebeetle esp32 board however there is already the "firebeetle32" configuration for that board.
DFRobot do not sell an esp32s3 board as far as I can see. 

If the "dfrobot_firebeetle2_esp32s3" configuration is meant to link to the [DRF0654 board](https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654), both the .json file on this repo and the arduino_pins.h file are incorrect which has been an issue for a while it seems - https://github.com/platformio/platform-espressif32/issues/644 and  https://github.com/lmarzen/esp32-weather-epd/issues/3

The current fix is fix is to use the "firebeetle32" configuration and set `platform = espressif32@5.0.0` in the .ini file and define all the pins manually. The "dfrobot_firebeetle2_esp32s3" configuration was added in release 5.1.0 and this changed the size of the flash and ram size which means it does not work for later versions. It would however be much nicer for it to work properly.

### Maybe how to fix?

- The [pins_ardunio.h](https://github.com/espressif/arduino-esp32/blob/master/variants/dfrobot_firebeetle2_esp32s3/pins_arduino.h) needs to be fixed in the the Arduino core and a correct version can be found on the DFRobot website In this [zip folder](https://download.dfrobot.top/firebeetle/DFRobot_FireBeetle_ESP32-E-0.2.0.zip)
- `"ldscript": "esp32s3_out.ld"` needs to be changed to `"ldscript": "esp32_out.ld"` in the .json file for "dfrobot_firebeetle2_esp32s3"
- `"mcu": "esp32s3",` also needs to be changed to `"mcu": "esp32",` 
- `"openocd_target": "esp32s3.cfg"` also needs to be changed to `"openocd_target": "esp-wroom-32.cfg"` 

I have tested this on my computer and everything seems to work properly with a few basic example sketches. However, I do not really understand what the rest of the .json does so not sure if anything else should be changed. Furthermore, I think the name of the configuration needs to be changed to dfrobot_firebeetle2_esp32-e however this seems complicated as i think other files would also need to be changed to link everything together

@valeros I am not too familiar with how platformio configurations or the arduino core work so could you offer some advice?

Thanks very much.

valeros, 2023-01-31:
Hi @ejagombar 

> The board for the configuration "dfrobot_firebeetle2_esp32s3" does not seem to exist. The .json file links to the webpage for the firebeetle esp32 board however there is already the "firebeetle32" configuration for that board.
DFRobot do not sell an esp32s3 board as far as I can see.

Maybe DFRobot are not selling the S3-based boards yet. The `dfrobot_firebeetle2_esp32s3` board and its variant were originally taken from https://github.com/espressif/arduino-esp32/pull/6890 and it's definately not the `DRF0654` board, so the link in the .json file points to a wrong product page.  

Your workaround is fine until a proper variant for DRF0654 is added to the ESP32 repo.

lmarzen, 2023-02-11:
Here is another work around, copied from https://github.com/platformio/platform-espressif32/issues/644...
In order to upload code to the Firebeetle 2 ESP32-E using the latest version of espressif32, you must override the flash size settings of the firebeetle32 board configuration. See my platformio.ini file below.

```
[env:firebeetle32]
platform = espressif32
board = firebeetle32
framework = arduino

; override default settings to enable compatibility with Firebeetle 2 ESP32-E
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
```