platformio/platform-espressif32 #1004: Updating to Espressif32@6.0.0 will break upload

jenschr, 2023-01-17:
Two of my students updated dependencies in Platformio today and they both get the error during compilation:

```
Compiling .pio/build/adafruit_feather_esp32s3_nopsram/src/main.cpp.o
*** [.pio/build/adafruit_feather_esp32s3_nopsram/partitions.bin] Source `partitions-4MB-tinyuf2.csv' not found, needed by target `.pio/build/adafruit_feather_esp32s3_nopsram/partitions.bin'.
```

This error occurs when the espressif32 tools are updated to version 6.0.0. 

This is confirmed using Adafruit Feather ESP32-S3 NoPSRAM (https://www.adafruit.com/product/5323) to test with. If you run into this issue using a different board, leave a message on this issue with the board name. 

A simple workaround is to force a lower version of espressif32 in the platformio.ini file:

`platform = espressif32@5.3.0`

This will make it all work as it should.

Example platformio.ini file for ESP32S3:

```
[env:adafruit_feather_esp32s3_nopsram]
platform = espressif32
board = adafruit_feather_esp32s3_nopsram
framework = arduino
monitor_speed = 115200
build_flags = -DARDUINO_USB_MODE=1
```

jenschr, 2023-01-17:
@Jason2866 Sure. It's just yet another thing that the students (and everybody else using the ESP32 S3) will have to learn.

The problem is however that that espressif32@6.0.0 breaks a behavior that has worked for many months. It selects the wrong file. That is the bug I'm reporting here.

Jason2866, 2023-01-17:
Fully understand. Before it was wrong too, and it was changed ;-) The mess here many boards are sold under the same name with different specs. 
With the S3 and OPI flash the situation has become worse. Even the Espressif Arduino guys needed some attempts to support it. Build bootloader correctly, boot flash mode and some other stuff. I had to dive deep in that to get my S3 with OPI correctly booting and working.
Did a few PRs....

closed 2023-01-17 by valeros

jenschr, 2023-01-22:
Yeah. This is a good fix as well. It would be really wise to roll this change out as quickly as possible. There's a host of boards coming out these days that are based on the S3. The WT32-SC01 PLUS is really popular with hackers since you get a good TFT with touch. These will all hit this problem, so if a 6.0.1 fix is out that solves this permanently, they'll get that instead of the 6.0.0.

As mentioned in the linked issue, but these two lines should be added as default platformio.ini for S3 also:

```
monitor_speed = 115200
build_flags = -DARDUINO_USB_MODE=1
```

The first just sets the expected baud rate for the module, but the other selects the correct USB Upload mode. These boards can use their internal USB as either upload or OTG Host. Setting -DARDUINO_USB_MODE=1 will make uploading the default. It's 99% more likely that someone is uploading firmware than using USB Host. And - if you're using USB Host, you probably want to add some more config anyway.

So if this line is added to the default platformio.ini for this board, it'll be a lot less support requests in the coming months.

Jason2866, 2023-01-22:
@jenschr  as you wrote many new boards with a S3 are already there and many more will come. It will be nearly impossible to add every board. The planned and already in the works Platformio Boards Configurator will solve this problem.

jenschr, 2023-01-22:
@Jason2866 This thread is specifically about the Adafruit Feather S3 with no PSRAM. I have a board similar to that, but I also have other boards that need this fix. I'd love to hear which other S3 with no PSRAM boards that this won't work for?

Jason2866, 2023-01-22:
I understood this for a standard for all S3 devices. The setting should only be used when no extra USB serial converter chip is on the board.

jenschr, 2023-01-22:
@Jason2866 I'm quite sure that all Adafruit Feather ESP32 S3 boards will use the internal USB for programming, but obviously Adafruit should tell that. It's sort of the main selling point of the S2 & S3 that you only need a USB chip if you want to add USB Host. But sure... I'll leave it at that and then we can just watch the issue list to see if I'm right? ;-)

Jason2866, 2023-01-22:
Looked at Adafruit site. Yes it looks like all S3 boards from Adafruit have no Serial USB chip.
So the USB setting you suggested is indeed the one which has to be in the boards.json!

commit a9ad5ac3 (2023-01-17): Fix partition file names for Feather S3 and QT Py S3 without PSRAM

Resolve #1004