platformio/platform-espressif32 #1006: [WinError 5] Access is denied on charset_normalizer

MarkoHammersen, 2023-01-19:
Greetings

when I try to build a PlatformIO project on WIN11 I get the error below.
Any idea what might be causing this issue?
Your help is greatly appreciated.
Regards
Marko

Processing myProject (platform: espressif32@5.3.0; framework: espidf; board: myProject)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
[extra_script.py] I am your extra script.
create working open api file with correct version....
tricore-objcopy: '0.0.0': No such file
tricore-objcopy:./scripts/update-version-in-openapi-yaml.sh: File format not recognized
version.bat v1.1
'patch' is not recognized as an internal or external command,
operable program or batch file.
Attempting to install intelHex...
b'Requirement already satisfied: intelhex in c:\\users\\mhammer3\\.platformio\\penv\\lib\\site-packages (2.3.0)\r\n'
Attempting to generate the iqs5xx firmware and customSettings header file...
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Installing ESP-IDF's Python dependencies
Collecting cryptography<35.0.0,>=2.1.4
  Using cached cryptography-3.4.8-cp36-abi3-win_amd64.whl (1.6 MB)
Collecting future>=0.15.2
  Using cached future-0.18.3-py3-none-any.whl
Collecting kconfiglib==13.7.1
  Using cached kconfiglib-13.7.1-py2.py3-none-any.whl (145 kB)
Collecting idf-component-manager~=1.0
  Using cached idf_component_manager-1.2.2-py3-none-any.whl (116 kB)
Collecting cffi>=1.12
  Using cached cffi-1.15.1-cp39-cp39-win_amd64.whl (179 kB)
Collecting colorama
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting click
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting cachecontrol[filecache]>0.12.6
  Using cached CacheControl-0.12.11-py2.py3-none-any.whl (21 kB)
Collecting requests<3
  Using cached requests-2.28.2-py3-none-any.whl (62 kB)
Collecting schema
  Using cached schema-0.7.5-py2.py3-none-any.whl (17 kB)
Collecting six
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting pyyaml>5.2
  Using cached PyYAML-6.0-cp39-cp39-win_amd64.whl (151 kB)
Collecting tqdm<5
  Using cached tqdm-4.64.1-py2.py3-none-any.whl (78 kB)
Collecting contextlib2>0.6.0
  Using cached contextlib2-21.6.0-py2.py3-none-any.whl (13 kB)
Collecting requests-toolbelt
  Using cached requests_toolbelt-0.10.1-py2.py3-none-any.whl (54 kB)
Collecting packaging
  Using cached packaging-23.0-py3-none-any.whl (42 kB)
Collecting requests-file
  Using cached requests_file-1.5.1-py2.py3-none-any.whl (3.7 kB)
Collecting msgpack>=0.5.2
  Using cached msgpack-1.0.4-cp39-cp39-win_amd64.whl (62 kB)
Collecting lockfile>=0.9
  Using cached lockfile-0.12.2-py2.py3-none-any.whl (13 kB)
Collecting pycparser
  Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
Collecting urllib3<1.27,>=1.21.1
  Using cached urllib3-1.26.14-py2.py3-none-any.whl (140 kB)
Collecting idna<4,>=2.5
  Using cached idna-3.4-py3-none-any.whl (61 kB)
Collecting charset-normalizer<4,>=2
  Using cached charset_normalizer-3.0.1-cp39-cp39-win_amd64.whl (96 kB)
Collecting certifi>=2017.4.17
  Using cached certifi-2022.12.7-py3-none-any.whl (155 kB)
Installing collected packages: msgpack, lockfile, charset-normalizer, urllib3, six, pyyaml, pycparser, packaging, kconfiglib, idna, future, contextlib2, colorama, certifi, tqdm, schema, requests, click, cffi, requests-toolbelt, requests-file, cryptography, cachecontrol, idf-component-manager
  Attempting uninstall: msgpack
    Found existing installation: msgpack 1.0.4
    Uninstalling msgpack-1.0.4:
      Successfully uninstalled msgpack-1.0.4
  Attempting uninstall: lockfile
    Found existing installation: lockfile 0.12.2
    Uninstalling lockfile-0.12.2:
      Successfully uninstalled lockfile-0.12.2
  Attempting uninstall: charset-normalizer
    Found existing installation: charset-normalizer 3.0.1
    Uninstalling charset-normalizer-3.0.1:
      Successfully uninstalled charset-normalizer-3.0.1
**ERROR: Could not install packages due to an OSError: [WinError 5] Access is denied: 'C:\\Users\\mhammer3\\.platformio\\penv\\Lib\\site-packages\\~0arset_normalizer\\md.cp39-win_amd64.pyd'
Check the permissions.**

*** Error 1
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "C:/Users/mhammer3/Documents/Kunden/myCustomer/main/.pio/build/myProject/CMakeFiles/CMakeOutput.log".

**fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/mhammer3/.platformio/packages/framework-espidf/tools/cmake/build.cmake:452 (message):
  C:\Users\mhammer3\.platformio\penv\Scripts\python.exe: Error while finding
  module specification for 'idf_component_manager.prepare_components'
  (ModuleNotFoundError: No module named 'idf_component_manager')**

Call Stack (most recent call first):
  C:/Users/mhammer3/.platformio/packages/framework-espidf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:4 (project)

============================================================================================ [FAILED] Took 6.90 seconds ============================================================================================

Environment     Status    Duration
--------------  --------  ------------
myProject  FAILED    00:00:06.902
====================================================================================== 1 failed, 0 succeeded in 00:00:06.902 ======================================================================================

MarkoHammersen, 2023-01-19:
I ran an pio

closed 2023-01-19 by MarkoHammersen

Caesar1113, 2023-01-29:
i got exactly same issue, i tried to clean it but the solution failed to me

Sebight, 2023-06-03:
Same thing is happening to me right now. `pio run -t clean` does not resolve the issue. Any idea how to fix this?