platformio/platform-espressif32 #1007: With v6 (ESP-IDF v5): `ImportError: cannot import name 'rest_of_line' from 'pyparsing'`

benwills, 2023-01-20:
I just upgraded to platform-espressif32 v6 the other day. Everything was working and compiling+uploading fine before the upgrade.

I refactored code based on compile errors I was expecting from the switch to ESP-IDF v5, but am still getting compile errors that seem that they may be related to platform-espressif32.

I'm now stuck on the following error, which seems to be part of the build process:

> Traceback (most recent call last):
>   File "/home/o/.platformio/packages/framework-espidf/tools/ldgen/ldgen.py", line 16, in <module>
>     from ldgen.entity import EntityDB
>   File "/home/o/.platformio/packages/framework-espidf/tools/ldgen/ldgen/entity.py", line 12, in <module>
>     from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
> ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/home/o/.platformio/penv/lib/python3.10/site-packages/pyparsing.py)

This error always occurs near the end of the build process.

----
### Steps I've taken:

Looking through similar errors, some people are able to work around it by doing a full clean, then building. That's not working for me. (Running Clean All, then running Build, both from the PlatformIO interface for my specific boards still fails, as seen below)

I have also completely uninstalled and deleted the platform-espressif32 files in the .pio directory, before doing a reinstall of platform-espressif32. But that also seems to not be working.

----
### System Configuration:

VSCode:
> Version: 1.74.3
> Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
> Date: 2023-01-09T16:57:40.428Z
> Electron: 19.1.8
> Chromium: 102.0.5005.167
> Node.js: 16.14.2
> V8: 10.2.154.15-electron.0
> OS: Linux x64 5.15.0-56-generic snap
> Sandboxed: No

PlatformIO:
> $ pio upgrade
> You're up-to-date!
> PlatformIO 6.1.5 is currently the newest version available.

platform-espressif32
> v6.0.0

OS
> Ubuntu 22.04 (xubuntu)
> $uname -a:
>> Linux xub 5.15.0-56-generic #62-Ubuntu SMP Tue Nov 22 19:54:14 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

----
### Output

Two ways of getting the error. First, is when I don't run 'Clean All' before 'Build', and the second is when I do run 'Clean All' before 'Build'

----
### Running 'Build' _without_ first running 'Clean All'

> Processing esp-wrover-kit (board: esp-wrover-kit; platform: espressif32; framework: espidf)
> 
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
> PLATFORM: Espressif 32 (6.0.0) > Espressif ESP-WROVER-KIT
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES: 
>  - framework-espidf @ 3.50000.0 (5.0.0) 
>  - tool-cmake @ 3.16.4 
>  - tool-esptoolpy @ 1.40400.0 (4.4.0) 
>  - tool-ninja @ 1.7.1 
>  - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
[... 272 lines cut ...]
...trucated to fit 65,536 characters...

> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/pcnt_periph.o
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/rmt_periph.o
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/sdmmc_periph.o
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/spi_periph.o
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/timer_periph.o
> Traceback (most recent call last):
>   File "/home/o/.platformio/packages/framework-espidf/tools/ldgen/ldgen.py", line 16, in <module>
>     from ldgen.entity import EntityDB
>   File "/home/o/.platformio/packages/framework-espidf/tools/ldgen/ldgen/entity.py", line 12, in <module>
>     from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
> ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/home/o/.platformio/penv/lib/python3.10/site-packages/pyparsing.py)
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/sdio_slave_periph.o
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/rtc_io_periph.o
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/touch_sensor_periph.o
> Compiling .pio/build/esp-wrover-kit/bootloader/soc/esp32/uart_periph.o
> Generating partitions .pio/build/esp-wrover-kit/partitions.bin
> Generating an empty partition .pio/build/esp-wrover-kit/ota_data_initial.bin
> *** [.pio/build/esp-wrover-kit/sections.ld] Error 1
> 
> [FAILED] Took 28.25 seconds
> 
> Environment     Status    Duration
> esp-wrover-kit  FAILED    00:00:28.254

benwills, 2023-01-20:
A similar issue was raised about a year ago in the esp-idf repo here: https://github.com/espressif/esp-idf/issues/8489

The ["resolution"](https://github.com/espressif/esp-idf/issues/8489#issuecomment-1289681866) was to run `idf.py fullclean` before a build.

I'm assuming that's what 'Clean All' does.

valeros, 2023-01-20:
Hi @benwills ! There is a chance that the pyparsing package is broken for some reason. Could you please open a new terminal window directly in the IDE:
[image]

And run `python -m pip install -U pyparsing --force-reinstall`

benwills, 2023-01-20:
Hey @valeros. Thank you! That seems to have fixed it.

I really appreciate the help on that.

closed 2023-01-20 by benwills