platformio/platform-espressif32 #1009: [Feature Request] - Board Support for LilyGo Esp32 T-Display-S3 8Mo

utopios, 2023-01-23:


rolupusoru, 2023-02-05:
Created draft version for this board in attachment. There is inconsistency between advertised flash memory size by vendor and the chip used in the schematic documentation. According to schematic external flash chip is a W25Q32 from Winbond with a capacity of 4MB (32Mbit).  This draft assumes that the flash size is 4MB until actual board arrives for further inspection.

To test this draft you must copy **lilygo-t-dysplay-esp32-s3.json** to _**%HOMEPATH%\platform.io\packages\espressif32\boards**_ . It is assumed that Platform.io is located in default installation folder. **partitions_2MB.csv** to your project directory. Restart VSCode and board will show up in the list. 

Add following line to _**platform.ini**_ to use included partition file.

`board_build.partitions = partitions_2MB.csv`

[lilygo_t_dysplay_s3.zip](https://github.com/platformio/platform-espressif32/files/10613953/lilygo_t_dysplay_s3.zip)

Eistee82, 2023-02-07:
https://github.com/platformio/platform-espressif32/pull/1015

rolupusoru, 2023-02-08:
This particular board comes with an ESP32-S3R8 chip  that has  8MB PSRAM, but only the first 4MB are in the address space of MMU by default due to address space limitation constraints. To go above this limit you need to use _**himem**_ bank switching. [highmem_api](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/himem.html)

An example of full 8MB PSRAM use can be found here:  [himem_example](https://github.com/espressif/esp-idf/tree/master/examples/system/himem)

**Note**: ESP32-S3 has 512KB RAM. PltaformIO advertises all S3 boards  with 320KB RAM. Possible 
copy mistake from ESP32-S2?

commit b94ae136 (2023-02-27): Update Arduino core to v2.0.7

- New boards (resolves #915, resolves #1009)
- Updated IDF to v4.4.4 for mixed projects
closed 2023-02-27 by valeros