platformio/platform-espressif32 #1014: Documentation for Adafruit QT Py ESP32-C3 says no internal debugger, yet Windows Device Manager reports USB JTAG

RufusVS, 2023-01-26:
The PlatformIO documentation page for the Adafruit QT Py ESP32-C3 says an external debugger is required, yet when I plug in my board, a USB JTAG serial device is reported.   Is this a documentation error or is the JTAG serial port unusable?_

Jason2866, 2023-01-26:
The inbuilt jtag device does work. No need for an extra external jtag debugging device.

RufusVS, 2023-01-26:
@Jason2866 Thanks, that what I thought.  Now my problem is: How do I configure VSCODE/PlatformIO/GDB/OpenOCD/Expressif to recognize the internal jtag and use it?

Jason2866, 2023-01-26:
An ocd debug env example
```
[env:debug32c3cdc-ocd]
build_type              = debug
board                   = adafruit_qtpy_esp32c3
debug_tool              = esp-builtin
upload_protocol         = esp-builtin
debug_init_break        = tbreak setup
```

RufusVS, 2023-01-29:
It turns out to be an incorrect driver that windows chooses.    You can get the tool installer [here.](https://dl.espressif.com/dl/esp-idf/?idf=4.4)   Then make sure to select: WinUSB support for JTAG.   After the driver is installed, go to device manager and select the JTAG device (under USB Serial Bus Devices).   Right-click and select Update Driver/Search automatically.  It should find the new driver called USB JTAG debug unit or something like that.  Install the new driver.  When I checked Properties/Driver I got this:
[image]

closed 2023-01-29 by RufusVS