platformio/platform-espressif32 #1018: Please add WiFi Kit 32 (V3) to board

blue0ceanby, 2023-01-29:


blue0ceanby, 2023-01-29:
Thank you, now I'm one step further.
Now I still have this error:

.pio/libdeps/heltec_wifi_kit_32_V3/U8g2/src/U8x8lib.cpp:45:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.

Gabone0001, 2023-01-29:
Search and install in platformio for U8g2 under Libraries

blue0ceanby, 2023-01-29:
Hello Gabone0001,
thanks for your support. Unfortunately the error message comes up:
Error: Unknown board ID 'heltec_wifi_kit_32_V3'

Gabone0001, 2023-01-29:
Did you copy the full code that i posted earlier including the links in platformio.ini ?
That should download the .git file for your board. Make sure that is a wifi kit 32 v3 and not another variant . You also have lora kit 32 v3 and a few more . 
Make sure that you have the git installed so it can download the file .

blue0ceanby, 2023-01-29:
ok, thanks to your tips I'm making good progress.
The build is already working. Thanks.
Now the download returns the error:
A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
***[Upload]Error 2

Do you have another idea?

valeros, 2023-01-31:
This board won't be added until this issue is resolved https://github.com/espressif/arduino-esp32/issues/7737 and a new release of ESP32 core is published.

blue0ceanby, 2023-02-03:
ok

Matty-hue, 2023-09-07:
@blue0ceanby Were you ever able to resolve this? I am having the same issue where I receive the following error.  Note U8g2 is installed  .pio/libdeps/heltec_wifi_kit_32_V3/U8g2/src/U8x8lib.cpp:45:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.
*** [.pio\build\heltec_wifi_kit_32_V3\libabf\U8g2\U8x8lib.cpp.o] Error 1

Jason2866, 2023-09-07:
Without posting the Platformio setup, no help possible.
Even better push the project to github and share the link.

Matty-hue, 2023-09-13:
@Jason2866 > Without posting the Platformio setup, no help possible. Even better push the project to github and share the link.

I've pushed to GH as you mentioned. https://github.com/Matty-hue/firstLight

Jason2866, 2023-09-14:
@Matty-hue Pin your platform. Without it is using what is currently installed. Probably an old one in your case

This one, compiles successfully
```
[env:heltec_wifi_kit_32_V3]
platform = espressif32 @ ^6.4.0
board = heltec_wifi_kit_32_V3
framework = arduino
```

```
Processing heltec_wifi_kit_32_V3 (platform: espressif32 @ ^6.4.0; board: heltec_wifi_kit_32_V3; framework: arduino)
--------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_kit_32_V3.html
PLATFORM: Espressif 32 (6.4.0) > Heltec WiFi Kit 32 (V3)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/heltec_wifi_kit_32_V3/src/main.cpp.o
Building .pio/build/heltec_wifi_kit_32_V3/bootloader.bin
Generating partitions .pio/build/heltec_wifi_kit_32_V3/partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/Print.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/USB.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/USBCDC.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/USBMSC.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/WString.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/base64.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-cpu.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-dac.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-gpio.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-i2c-slave.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-i2c.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-ledc.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-matrix.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-misc.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-psram.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-rgb-led.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-spi.c.o
[... 3 lines cut ...]
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/main.cpp.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/heltec_wifi_kit_32_V3/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/heltec_wifi_kit_32_V3/libFrameworkArduino.a
Indexing .pio/build/heltec_wifi_kit_32_V3/libFrameworkArduino.a
Linking .pio/build/heltec_wifi_kit_32_V3/firmware.elf
Retrieving maximum program size .pio/build/heltec_wifi_kit_32_V3/firmware.elf
Checking size .pio/build/heltec_wifi_kit_32_V3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.7% (used 18552 bytes from 327680 bytes)
Flash: [=         ]   8.0% (used 268129 bytes from 3342336 bytes)
Building .pio/build/heltec_wifi_kit_32_V3/firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
===================================== [SUCCESS] Took 4.32 seconds =====================================
 *  Terminal will be reused by tasks, press any key to close it. 
 ```

Matty-hue, 2023-09-14:
@Jason2866 that worked! thank you