platformio/platform-espressif32 #1021: add_definitions(-DCONFIG_IDF_TARGET_ESP32xxx=true) required in CMakeListsPrivate.txt for recent espressif versions

j4m3s, 2023-02-05:


Jason2866, 2023-02-05:
Probably not fixed soon. https://github.com/platformio/platform-espressif32/issues/994#issuecomment-1377150195

valeros, 2023-02-06:
CLion is currently unsupported as development environment for ESP-IDF, see https://docs.platformio.org/en/latest/integration/ide/clion.html#clion-with-esp-idf for more details.

closed 2023-02-06 by valeros

j4m3s, 2023-02-06:
@valeros I think it's worth strengthening that paragraph in the docs, or better-still, putting a bold note at the top of that page to say that CLion isn't supported.  I infer that paragraph to have such strong implications as to say that "CLion is currently unsupported as development environment for ESP-IDF" as you put it here.