platformio/platform-espressif32 #1023: It burns but don't work after updating platform (Espressif ESP 32 C3)

brunofgc, 2023-02-07:
Hi there,

After updating platform and libs, my project stopped working.
I noticed that it burns the microcontroller much quicker than it used to burn.
It starts slow as before, and after arround 20% it get accelerating and being fast.

When I reset the device (m5stamp c3) it doesn't work as espected.

I'm sure that there is no problem with my code itself because I didn't change it before updating.

It was working, I noticed the new release, just updated it and it stopped working.

[image]

-- Enviroment

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (6.0.1+sha.764527e) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries

Jason2866, 2023-02-08:
Works with every code tried without any issue.

valeros, 2023-02-08:
Hi @brunofgc ! What was the previous version you used? Does your code work in the Arduino IDE? Also please share your `platformio.ini` file here.

brunofgc, 2023-02-09:
Hi @valeros and @Jason2866 

This is my platformio.ini

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-c3-devkitm-1]
platform = https://github.com/platformio/platform-espressif32.git
;platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
lib_deps = 
	knolleary/PubSubClient@^2.8
	h2zero/NimBLE-Arduino@^1.4.0
	marian-craciunescu/ESP32Ping@^1.7
	plerup/EspSoftwareSerial@^6.16.1
	vshymanskyy/TinyGSM@^0.11.5
	fastled/FastLED@^3.5.0
	makuna/NeoPixelBus@^2.7.0
	adafruit/Adafruit NeoPixel@^1.10.5
`

As you can see, I tried both:
platform = https://github.com/platformio/platform-espressif32.git
;platform = espressif32

But none of them worked.

It was working fine before I updated.

I didn't try it on Arduino's IDE.

I think something went wrong while updating because it was working fine with no need for changing. Just tried to keep it up to date.

Should I try on Arduino's IDE?

valeros, 2023-02-09:
> Should I try on Arduino's IDE?

It'd be great, so that we will know there are no issues with the framework itself.

brunofgc, 2023-02-11:
@valeros Hi dear friend. It's not compiling on Arduino 2.0 IDE.

[image]

I installed all the dependencies, all the libraries. It was working, and saying "not found lib blablabla" I installed "blablabla" lib... One by one. It was compiling. Now it's stuck as you can see.

Reset the whole computer and tried again. I got the same result.

brunofgc, 2023-02-11:
@valeros It worked!
It compilled and flashed properly.

Using Arduino IDE.

What should I do to have it working again with VSCode and PlatformIO?

It seems to be something with that specific release.

valeros, 2023-02-13:
@brunofgc Please compile your project using the Arduino IDE in verbose mode and share here the log file.

brunofgc, 2023-02-13:
@valeros 

Here we have the outputs -> https://codefile.io/f/CYSS0JjpHmSkXdgTQjC3

valeros, 2023-02-14:
@brunofgc thanks for the logs. Could you also describe in more detail what exactly doesn't work as expected?

Jason2866, 2023-02-14:
@brunofgc  The Arduino IDE compile log shows that parts of the project are stored on One Drive. What happens if everything is stored local when compiling with VSC?

brunofgc, 2023-02-14:
@valeros @Jason2866 
Hi there.

All off the project was writen in VSCode with the same conditions I have right now. I.E. same path/folders/location in HDD.

It was working okay.

I didn't want to change anything in the code from previous compilation, but I thought "update the libs" would be a good idea.

My procedure was going to update them. I updated the platform and it stopped working.

I don't recall if I updated any other lib.

Suddenly it stopped working. It Compiles! But the firmware doesn't start, doesn't work.

Jason2866, 2023-02-14:
You have used a different (older) core in Arduino. In Arduino IDE v2.0.3 is used
Latest is v2.0.6. You have used with VSC.
So you compared core v2.0.3 Arduino IDE with core 2.0.6 VSC.
Please update in Arduino IDE to core 2.0.6 and test if it is still working there.

brunofgc, 2023-02-14:
@Jason2866 I'm afraid of doing that right now, because I'm covering a big project and updating the firmware constantly. I can't be unable to compile.

I'll need to wait a bit, since I have a way to keep working.

closed 2023-03-25 by stale[bot]