platformio/platform-espressif32 #1024: apply git patch to PIO's esp-idf

marchingband, 2023-02-08:
I have a git patch file that allows esp32s3 to act as a usb midi host.
It is taking forever for this patch to make its way into an esp-idf release.
Is there a way to use this patch file with PIO CORE?
I have a git repo setup already with a fork of the esp-idf, and the patch applied
thank you!

valeros, 2023-02-09:
Hi @marchingband ! If your fork is based on the stable IDF v5.0 then you can simply use the [platform_packages](https://docs.platformio.org/en/latest/projectconf/sections/env/options/platform/platform_packages.html) option (your repo must contain the `package.json` file from the IDF package installed by PlatformIO). Alternatively, you can apply you dinamically patch to the installed IDF package via an extra script as described [here](https://docs.platformio.org/en/latest/scripting/examples/override_package_files.html).

closed 2023-02-09 by valeros

marchingband, 2023-02-10:
@valeros Thank you! That's great. Do I need to modify package.json ?
Perhaps replace repository.url with my own for example?

Jason2866, 2023-02-10:
@marchingband  You can modify like you want or not. Just a valid one has to be there.

marchingband, 2023-03-03:
Perhaps I misunderstood your response.
I have this .ini
```
[env:esp32-s3-devkitc-1]
platform = espressif32
framework = espidf
platform_packages = framework-espidf @ https://github.com/marchingband/mainframe-idf.git
board = esp32-s3-devkitc-1
```
but `pio run` produces
```
KeyError: 'framework-espressif32':
  File "/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/platformio/builder/main.py", line 171:
    env.LoadPioPlatform()
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/platformio/builder/tools/pioplatform.py", line 75:
    type_ = p.get_package_type(pkg.metadata.name)
  File "/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/platformio/platform/base.py", line 175:
    return self.packages[name].get("type")
```
Shouldn't pio automatically download the repo, etc? Do I need to install esp-idf manually?

Jason2866, 2023-03-03:
Your fork is IDF 4.4.2. The actual platformio version is for IDF 5.0.1. You cant mix the versions.
This should match

```
[env:esp32-s3-devkitc-1]
platform = espressif32 @ 5.2.0
board = esp32-s3-devkitc-1
framework = espidf
platform_packages = framework-espidf @ https://github.com/marchingband/mainframe-idf.git
```

marchingband, 2023-03-03:
Ah good catch, unfortunately I get the same error with those changes.

Jason2866, 2023-03-03:
Tried, works. Seems your local setup is screwed up.

marchingband, 2023-03-12:
failed with this odd error, any idea what this is about?
```
Compiling .pio/build/esp32-s3-devkitc-1/bootloader/bootloader_support/src/bootloader_random.o
Indexing .pio/build/esp32-s3-devkitc-1/bootloader/esp-idf/main/libmain.a
Traceback (most recent call last):
  File "/Users/andy/.platformio/packages/framework-espidf/tools/ldgen/ldgen.py", line 16, in <module>
    from ldgen.entity import EntityDB
  File "/Users/andy/.platformio/packages/framework-espidf/tools/ldgen/ldgen/entity.py", line 12, in <module>
    from pyparsing import (Group, Literal, OneOrMore, ParseException, SkipTo, Suppress, White, Word, ZeroOrMore, alphas,
ImportError: cannot import name 'rest_of_line' from 'pyparsing' (/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/pyparsing.py)
Compiling .pio/build/esp32-s3-devkitc-1/bootloader/bootloader_support/src/bootloader_random_esp32s3.o
*** [.pio/build/esp32-s3-devkitc-1/sections.ld] Error 1
```

Jason2866, 2023-03-12:
Maybe Python v3.11 is not working with IDF. I would try Python v3.9 or v3.10

valeros, 2023-03-13:
@marchingband It seems like a version mismatch. What version of the dev-platform are you using?

marchingband, 2023-03-14:
@valeros platforms/espressif32/package.json says `version: 5.2.0`

marchingband, 2023-03-14:
@valeros i see its up to 6.1.0 now.
I have another important project on this machine pinned to 5.2.0, is it possible to mix and match?
When I ran `pio run` with this new project setup as instructed above, it installed the old esp-idf and broke the other project. It was easy to revert, but, how can I keep them separate?

valeros, 2023-03-14:
@marchingband Is there anything special in 5.2.0 that you pinned it? Do you need IDF v4?

marchingband, 2023-03-14:
@valeros yes both projects need v4, one of them needs the patch, the other does not

valeros, 2023-03-15:
@marchingband I should be possible to benefit from the latest version of the development platform and still use previous packages:
```ini
[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
platform_packages = 
    framework-espidf @ ~3.40404.0
    ; IDF v4 requires an old toolchain
    toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

[env:esp32dev_patched]
platform = espressif32
framework = espidf
board = esp32dev
platform_packages = 
    framework-espidf @ https://github.com/marchingband/mainframe-idf.git
    toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```

marchingband, 2023-05-06:
Still struggling with this.

```
[env:esp32-s3-devkitc-1]
platform = espressif32
framework = espidf
board = esp32dev
platform_packages = 
    framework-espidf @ https://github.com/marchingband/mainframe-idf.git
```

package.json:

```
{
  "name": "framework-espidf@mainframe",
  "version": "3.50000.0-patch",
  "description": "Patch of esp-idf 5.0 to add midi-usb.",
  "keywords": [
    "framework",
    "esp32",
    "esp32s2",
    "esp32s3",
    "espressif"
  ],
  "homepage": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/",
  "license": "Apache-2.0",
  "repository": {
    "type": "git",
    "url": "https://github.com/marchingband/mainframe-idf"
  }
}
```

problem 1: my patched esp-idf is version 5.0, do I have the wrong `toolchain-xtensa-esp32`?
problem 2: installing this repo always overwrites `/framework-espidf`, corrupting all my other pio projects, I cannot get it to instal in a new directory ... I would expect the directory to be `/framework-espidf@mainframe` or `/framework-espidf@3.50000.0-patch` based on package.json.
problem 3: `pio run` fails with

```
 - framework-espidf @ 3.50000.0-patch+sha.31e1b45 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
TypeError: can only concatenate str (not "NoneType") to str:
  File "/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/platformio/builder/main.py", line 187:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/andy/.platformio/platforms/espressif32/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 120:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/opt/homebrew/Cellar/platformio/6.1.6/libexec/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 331:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 660:
    return method(*args, **kw)
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/andy/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/andy/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1192:
    install_python_deps()
  File "/Users/andy/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1100:
    python_exe_path = get_python_exe()
  File "/Users/andy/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1170:
    env.subst("$PROJECT_CORE_DIR"), "penv", ".espidf-" + idf_version)
```

marchingband, 2023-05-06:
I have also taken the esp-idf installation directly from the machine where it is working (with the patch) and pushed it to git, but the same error gets thrown
`framework-espidf @ https://github.com/ultrapalace/mainframe-idf.git`

Jason2866, 2023-05-06:
You have to pin the version of the platform. Without doing this every project will use what is installed.

marchingband, 2023-05-06:
As a side note, there has been a challenge to determine information on my own. I don't know if I am missing some documentation around this stuff, but, for example, looking here `https://registry.platformio.org/tools/platformio/framework-espidf/versions` I dont see any useful information on what version of framework-espidf corresponds to the various esp-idf versions, so I am just guessing. Is there a github repo for framework-espidf? some other source of versioning info? thanks!

marchingband, 2023-05-06:
@Jason2866 how do I accomplish that?
My experience using `platform_packages` has been that each one gets its own folder and PIO is able to know what is what

Jason2866, 2023-05-06:
`platform = espressif32 @ 5.2.0` for example

Jason2866, 2023-05-06:
> As a side note, there has been a challenge to determine information on my own. I don't know if I am missing some documentation around this stuff, but, for example, looking here [...]

Thats really not user friendly. One possible way, look in platform.json from tagged commit of release.

marchingband, 2023-05-07:
> Thats really not user friendly. One possible way, look in platform.json from tagged commit of release.

Like i said I have been unable to locate the git repo for framework-espidf, can you let me know the URL, and how to locate the tagged commit of a release?

> `platform = espressif32 @ 5.2.0` for example and yes every change here will install the specified IDF again. There is no switching.

My question is how to install my own patched version, @valeros had said

> @marchingband I should be possible to benefit from the latest version of the development platform and still use previous packages:

I had tried using `platform = espressif32 @ 5.2.0` after your first suggesiton @Jason2866, but it caused a whole host of other problems, and using `platform = espressif32` has allowed me to use various esp-idf versions in parallel, so I am not sure that your advice is accurate here. If I am wrong, can you try to make this more clear? thanks