platformio/platform-espressif32 #1025: Adafruit Feather ESP32-S3 8MB No PSRAM - cannot upload

ruig25, 2023-02-10:
**platformio.ini**
platform = espressif32
board = adafruit_feather_esp32s3_nopsram
framework = espidf

**Problem**
Stucks on upload after switching to download mode

**Build log**
`Processing adafruit_feather_esp32s3_nopsram (platform: espressif32; board: adafruit_feather_esp32s3_nopsram; framework: 
espidf)
------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_feather_esp32s3_nopsram.html
PLATFORM: Espressif 32 (6.0.1+sha.764527e) > Adafruit Feather ESP32-S3 No PSRAM
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50000.0 (5.0.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1
 - toolchain-xtensa-esp32s3 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\adafruit_feather_esp32s3_nopsram\src\main.o
Generating LD script .pio\build\adafruit_feather_esp32s3_nopsram\memory.ld
Compiling .pio\build\adafruit_feather_esp32s3_nopsram\app_trace\app_trace.o
...
<a lot of *.o files>
...
Compiling .pio\build\adafruit_feather_esp32s3_nopsram\bootloader\soc\esp32s3\usb_otg_periph.o
Generating partitions .pio\build\adafruit_feather_esp32s3_nopsram\partitions.bin
Archiving .pio\build\adafruit_feather_esp32s3_nopsram\bootloader\esp-idf\soc\libsoc.a
Archiving .pio\build\adafruit_feather_esp32s3_nopsram\bootloader\esp-idf\micro-ecc\libmicro-ecc.a
Linking .pio\build\adafruit_feather_esp32s3_nopsram\bootloader.elf
Building .pio\build\adafruit_feather_esp32s3_nopsram\bootloader.bin
esptool.py v4.4
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Linking .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Retrieving maximum program size .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Checking size .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.8% (used 32140 bytes from 327680 bytes)
Flash: [=======   ]  69.1% (used 724641 bytes from 1048576 bytes)
Building .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v4.4
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
============================================ [SUCCESS] Took 164.80 seconds ============================================
 *  Terminal will be reused by tasks, press any key to close it. `

**Upload log**
`...
Linking .pio\build\adafruit_feather_esp32s3_nopsram\bootloader.elf
Building .pio\build\adafruit_feather_esp32s3_nopsram\bootloader.bin
esptool.py v4.4
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Linking .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Retrieving maximum program size .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Checking size .pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.8% (used 32140 bytes from 327680 bytes)
Flash: [=======   ]  69.1% (used 724641 bytes from 1048576 bytes)
Building .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v4.4
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM10
Forcing reset using 1200bps open/close on port COM10
Waiting for the new upload port...
Uploading .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v4.4
Serial port COM9 <--------- Stucks here`

**Menuconfig**
Flash SPI Mode = QIO

**Device manager**
Device before switching to download mode: VID:PID=0x239A:0x8113
Device after switching to download mode: VID:PID=0x303A:0x1001

**Arduino IDE**
ESP32 BLE Example compiled und uploaded without issues (QIO)

julian-w, 2023-02-12:
I have the same problem, with "espressif32@4.4.0" it is just working fine. esptool.py is then v3.3. Seems like a bug in esptool.py?

valeros, 2023-02-13:
Hi @ruig25 ! Please run the upload in verbose mode to see the exact step where it hangs.

ruig25, 2023-02-13:
> Hi @ruig25 ! Please run the upload in verbose mode to see the exact step where it hangs.

Hi!

**Log**

 *  Executing task: C:\Users\abc\.platformio\penv\Scripts\platformio.exe run --verbose --target upload --environment adafruit_feather_esp32s3_nopsram 

Processing adafruit_feather_esp32s3_nopsram (extends: common_env; board: adafruit_feather_esp32s3_nopsram; platform: espressif32; framework: espidf; upload_speed: 115200; monitor_speed: 115200; monitor_raw: True; build_flags: -std=gnu++14; build_unflags: -std=gnu++11; build_src_flags: -Wno-missing-field-initializers -Wno-error=switch)
----------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_feather_esp32s3_nopsram.html
PLATFORM: Espressif 32 (6.0.1) > Adafruit Feather ESP32-S3 No PSRAM
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50000.0 (5.0.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1
 - toolchain-xtensa-esp32s3 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
<lambda>(["checkprogsize"], [".pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf"])
MethodWrapper(["checkprogsize"], [".pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.8% (used 32140 bytes from 327680 bytes)
Flash: [=======   ]  69.1% (used 724641 bytes from 1048576 bytes)
.pio\build\adafruit_feather_esp32s3_nopsram\firmware.elf  :

section                                                         size         addr

.rtc.text                                                         16   1611653120

.rtc.force_fast                                                    0   1611653136

.rtc.data                                                         16   1342177280

.rtc_noinit                                                        0   1342177296

.rtc.force_slow                                                    0   1342177296

.iram0.vectors                                                  1027   1077362688

.iram0.text                                                    72895   1077363716

.dram0.dummy                                                   57600   1070104576

.dram0.data                                                    15628   1070162176

.noinit                                                            0   1070177804

.dram0.bss                                                     16512   1070177808

.flash.text                                                   500239   1107296288

.flash_rodata_dummy                                           524288   1006632992

.flash.appdesc                                                   256   1007157280

[... 117 lines cut ...]

.debug_ranges                                                  47560            0

.debug_line                                                  1914218            0

.debug_str                                                    401547            0

.debug_rnglists                                                 3059            0

.debug_line_str                                                13626            0

.debug_loclists                                                84053            0

Total                                                        9077661
<lambda>(["upload"], [".pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin"])
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
BeforeUpload(["upload"], [".pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin"])
TimeoutError: Could not automatically find serial port for the `Adafruit Feather ESP32-S3 No PSRAM` board based on the declared HWIDs=['239A:811B', '239A:011B', '239A:811C']
Auto-detected: COM5
Forcing reset using 1200bps open/close on port COM5
Waiting for the new upload port...
"C:\Users\abc\.platformio\penv\Scripts\python.exe" "C:\Users\abc\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32s3 --port "COM8" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 C:\UserData\abc\Documents\PlatformIO\Projects\sensors\.pio\build\adafruit_feather_esp32s3_nopsram\bootloader.bin 0x8000 C:\UserData\abc\Documents\PlatformIO\Projects\sensors\.pio\build\adafruit_feather_esp32s3_nopsram\partitions.bin 0x10000 .pio\build\adafruit_feather_esp32s3_nopsram\firmware.bin
esptool.py v4.4
Serial port COM8

ruig25, 2023-02-13:
> The first thing that catches the eye is a mess with serial ports. The auto-detected port is COM5 while PlatformIO is trying to use COM8. Which one is correct? Have you tried specifying the correct one using the [upload_port](https://docs.platformio.org/en/latest/projectconf/sections/env/options/upload/upload_port.html) option?

1) Device inserted. COM5 appears.
2) I suppose esptool switches the device to Enter ROM Bootloader Mode (COM5 disappears and new device COM8 appears)
3) esptool tries to upload to COM8...

Arduino IDE (as proof that device works).
Inserted device (COM5). Compile and upload... Device switched by software to ROM Bootloader Mode (COM5 -> COM8).
Upload complete. Everythin OK.

The device can also be switched to ROM Bootloader Mode manually.
Holding boot microswitch and pressing Reset once.

Jason2866, 2023-02-13:
Do you use tinyuf2? If not you can try `esp32-s3-devkitc-1` as board. It has 8MB QIO flash and no PSRAM too. It is not using tinyuf2. If this does work the issue is caused from tinyuf2 which switches the USB port.

ruig25, 2023-02-13:
> Do you use tinyuf2? If not you can try `esp32-s3-devkitc-1` as board. It has 8MB QIO flash and no PSRAM too. It is not using tinyuf2. If this does work the issue is caused from tinyuf2 which switches the USB port. Maybe this [espressif/arduino-esp32#7828](https://github.com/espressif/arduino-esp32/pull/7828) which will be in next release. For IDF the link in the PR is helpful. Since IDF version changed, maybe your code needs changes.

Tried `esp32-s3-devkitc-1` without success.

**Arduino IDE Upload log (as reference...)**
`Sketch uses 800265 bytes (38%) of program storage space. Maximum is 2097152 bytes.
Global variables use 44704 bytes (13%) of dynamic memory, leaving 282976 bytes for local variables. Maximum is 327680 bytes.
Forcing reset using 1200bps open/close on port COM5
PORTS {COM3, COM5, } / {COM3, COM8, } => {COM8, }
Found upload port: **COM8**
C:\Users\abc\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1/esptool.exe --chip esp32s3 --port COM8 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 C:\Users\abc\AppData\Local\Temp\arduino_build_754870/SimpleBleDevice.ino.bootloader.bin 0x8000 C:\Users\abc\AppData\Local\Temp\arduino_build_754870/SimpleBleDevice.ino.partitions.bin 0xe000 C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/partitions/boot_app0.bin 0x10000 C:\Users\abc\AppData\Local\Temp\arduino_build_754870/SimpleBleDevice.ino.bin 0x410000 C:\Users\abc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/variants/adafruit_feather_esp32s3_nopsram/tinyuf2.bin 
esptool.py v4.2.1
Serial port COM8
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:5d:bb:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
... etc`

ruig25, 2023-02-13:
Arduino settings (as reference...)

[image]

julian-w, 2023-02-13:
Maybe related to: https://github.com/espressif/esptool/issues/808

ruig25, 2023-02-13:
esptool PlatformIO vs Arduino command side by side comparision (full paths were cut to simplify the overview):

**PlatformIO**
esptool.py  --chip esp32s3 --port COM8 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 bootloader.bin 0x8000 partitions.bin 0x10000 firmware.bin

**Arduino IDE**
esptool.exe --chip esp32s3 --port COM8 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 SimpleBleDevice.ino.bin 0x410000 tinyuf2.bin

ruig25, 2023-02-13:
> Maybe related to: [espressif/esptool#808](https://github.com/espressif/esptool/issues/808)

**esptool.py v3.3-dev**
```
esptool.py --port COM8 --baud 115200 --chip esp32s3 chip_id
esptool.py v3.3-dev
Serial port COM8
Connecting...   
Chip is ESP32-S3      
Features: WiFi, BLE   
Crystal is 40MHz      
MAC: f4:12:fa:5d:bb:f0
Uploading stub...     
Running stub...
Stub running...
Warning: ESP32-S3 has no Chip ID. Reading MAC instead.
MAC: f4:12:fa:5d:bb:f0       
Hard resetting via RTS pin...
```

**esptool.py v4.4**
```
sptool.py --port COM8 --baud 115200 --chip esp32s3 chip_id
esptool.py v4.4
Serial port COM8
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:5d:bb:f0
Uploading stub...
Running stub...
```

valeros, 2023-02-13:
> esptool PlatformIO vs Arduino command side by side comparision (full paths were cut to simplify the overview):

It's not entirely accurate to compare the upload commands for Arduino and ESP-IDF. Anyway, according to https://github.com/espressif/esptool/issues/808, esptool v4.3 might help here, so could you downgrading the `tool-esptoolpy` package?

```ini
[env:...]
platform = espressif32
board = adafruit_feather_esp32s3_nopsram
framework = espidf

platform_packages = 
   tool-esptoolpy @ ~1.40300.0
```

Jason2866, 2023-02-13:
From the Arduino log. There is esptool v4.2.1 used. So try with this version. Hope I have remembered the syntax correctly for using a espressif release. Can't try at the moment
```
platform_packages = tool-esptoolpy @ https://github.com/espressif/esptool.git#tree/v4.2.1
```

Jason2866, 2023-02-13:
Can you please try if the released esptool v4.5 from today does work too?
Just made a package. You can use with
```
platform_packages = tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v4.5/esptool-4.5.zip
```

julian-w, 2023-02-13:
> @ruig25 Can you please try if the released esptool v4.5 from today does work too? Just made a package. You can use with
> 
> ```
> platform_packages = tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v4.5/esptool-4.5.zip
> ```

For me it is working 👍🏻

Jason2866, 2023-02-13:
Thx, for the feedback!

ruig25, 2023-02-13:
> @ruig25 Can you please try if the released esptool v4.5 from today does work too? Just made a package. You can use with
> 
> ```
> platform_packages = tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v4.5/esptool-4.5.zip
> ```

Works!

Jason2866, 2023-02-13:
@valeros  Can you release esptool.py v4.5 in platformio registry?
Arduino core 2.0.7 will use esptool.py v4.5. Espressif guys knew that v4.4 has a bug with the S3.

radimkarnis, 2023-02-14:
Thank you all for the feedback and confirmation that v4.5 solves the issues!

closed 2023-02-17 by valeros

Vitallyz, 2023-02-18:
Tried to use VSCode and PlatformIO today for the first time and got stuck with that issue! So glad it was solved just few days ago! Thanks a lot!

Vitallyz, 2023-02-19:
There ie another issue, I believe realted to that one, when I try to upload some projects (not any project) to ESP32-S3 Feather, during upload I get this error (every time on a different stage of uploading, and it could eventually upload, but only like 1 out of 10 trials):

`Successfully created esp32s3 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM11
Forcing reset using 1200bps open/close on port COM11
Waiting for the new upload port...
Uploading .pio\build\adafruit_feather_esp32s3\firmware.bin
esptool.py v4.5
Serial port COM11
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:59:b4:fc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x002d0000 to 0x002f8fff...
Flash will be erased from 0x00010000 to 0x00057fff...
Compressed 22624 bytes to 14246...
Writing at 0x00000000... (100 %)
Traceback (most recent call last):
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\__init__.py", line 1031, in _main
    main()
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\__init__.py", line 831, in main
    operation_func(esp, args)
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\cmds.py", line 580, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\loader.py", line 1030, in flash_defl_block
    self.check_command(
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\loader.py", line 404, in command
    p = self.read()
  File "C:\Users\vital\.platformio\packages\tool-esptoolpy@src-28ff350b3469df913de33039996646bc\esptool\loader.py", line 337, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.
*** [upload] Error 2`  

To solve that, I changed esptool version to 4.3:

`platform_packages = tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v4.3/esptool-4.3.zip`

ruig25, 2023-02-20:
> There ie another issue, I believe realted to that one, when I try to upload some projects (not any project) to ESP32-S3 Feather, during upload I get this error (every time on a different stage of uploading, and it could eventually upload, but only like 1 out of 10 trials):

I observed the same behaviour on one dev PC.
Sometimes upload finishes with error... sometimes not.
It doesn't happen on the other dev PC.

Vitallyz, 2023-02-20:
> I observed the same behaviour on one dev PC. Sometimes upload finishes with error... sometimes not. It doesn't happen on the other dev PC.

Yes, I also felt like it is hardware dependent. I tried with 3 different USB cables and the shortest did the best and failed at the later stages compared to longer cables, but upload would eventually fail for the short cable as well. 

However, after changing back to version 4.3 all cables worked well, with no upload issues at all.

commit 90130b69 (2023-02-17): Update esptoolpy to v4.5.0

Resolves #1025