platformio/platform-espressif32 #1028: breaks bootloader file name ..sdk\esp32\bin\bootloader__40m.bin

voklav, 2023-02-13:
Hello everyone.
I'm new here and I am sorry if I do something wrong, but I think there is a problem with the configuration of the bootloader file name in the esp32 module.

Fresh delivered [Freematics ONE+](https://freematics.com/products/freematics-one-plus-model-b/)
Properly configured environment (vscode + platformIO)
I am following the default steps.
Build (With no error)
Upload (With this error)

`Auto-detected: COM4
Uploading .pio\build\esp32dev\firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\USERNAME\\.platformio\\packages\\framework-arduinoespressif32\\tools\\sdk\\esp32\\bin\\bootloader__40m.bin'
*** [upload] Error 2`

the platformio.ini is the default one from [freematic GitHub repo](https://github.com/stanleyhuangyc/Freematics/blob/8ba45a065b26fbcca892d5b109e003718e66ec1d/firmware_v5/telelogger/platformio.ini)

Also I'm trying to change the version (I'm not sure if this is the right way, but it didn't work anyway)
Change
`platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.5`
to
`platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.6`

Any suggestion what to do ?

Jason2866, 2023-02-13:
This is a bad idea
```
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```
use
```
platform = espressif32 @ 5.3.0
```

valeros, 2023-02-13:
Hi @voklav ! The branch `https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream` is no longer maintained. The maintainers of the Freematics repo need to update their PlatformIO config files according to the latest changes in the development platform, otherwise you won't be able to use the latest Arduino cores.

closed 2023-02-13 by valeros