platformio/platform-espressif32 #1029: ESP-IDF 5...I don't get it?

nicklasb, 2023-02-13:
Hi!

I now think I'd tried everything, and somewhere here someone said that platform-espressif32 would install ESP-IDF version 5, however, it doesn't seem to do that. 
Even though framework-espidf @ 3.50000.0 (5.0.0) seems to claim be 5.0.0, and says so in the version.txt, it still seems to be 4.x when looking at the actual code. esp-now, for example, still contain the old callback definition. 

Have I completely misunderstood something? I am thoroughly confused... :-)
 
Kind regards, Nicklas

valeros, 2023-02-14:
Hi @nicklasb ! We use packages directly from the IDF repository here https://github.com/espressif/esp-idf/releases/tag/v5.0. Could you please elaborate what old callback definitions exactly are not consistent?

valeros, 2023-02-14:
> Why would a breaking change in the release branch not be part of the tagged version?

Honestly, I have no idea, but it seems `release/v5.0` is a just prerelease branch:
[image]

I guess not everything from that branch may end up in the stable release.

Anyway, I've just compared `https://github.com/espressif/esp-idf/blob/v5.0/components/esp_wifi/include/esp_now.h` and `esp_now.h` from 3.50000.0 (5.0.0) and they are exactly the same, so doesn't seem like a PlatformIO issue to me.

nicklasb, 2023-02-14:
Odd communication, I'd say, on github the pre-release categorization isn't visible. 
I agree, this is not on Platform IO, closing. 

Thanks for you time!

closed 2023-02-14 by nicklasb