platformio/platform-espressif32 #1030: Configuring incomplete when platformio=espidf,arduino

swortain, 2023-02-14:
I have updated all to the latest version and reopen the IDE, build the project.
then I got these message:

    Processing ESP32-S3 (platform: espressif32; board: esp32-s3-devkitc-1; framework: espidf, arduino)
    -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
    CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
    PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
    HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
    DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
    PACKAGES:
    - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
    - framework-espidf @ 3.40403.0 (4.4.3)
    - tool-cmake @ 3.16.4
    - tool-esptoolpy @ 1.40400.0 (4.4.0)
    - tool-idf @ 1.0.1
    - tool-mconf @ 1.4060000.20190628 (406.0.0)
    - tool-ninja @ 1.9.0
    - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
    - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
    - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
    Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
    Reading CMake configuration...
    -- Building ESP-IDF components for target esp32s3
    -- Project sdkconfig file C:/Users/swortain/OneDrive/PlatformIO/220319_ESP32S3_keyboard_value/sdkconfig.ESP32-S3
    -- App "220319_ESP32S3_keyboard_value" version: fb1f22d-dirty
    -- Adding linker script C:/Users/swortain/OneDrive/PlatformIO/220319_ESP32S3_keyboard_value/.pio/build/ESP32-S3/esp-idf/esp_system/ld/memory.ld
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/esp_system/ld/esp32s3/sections.ld.in
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib-time.ld
    -- Adding linker script C:/Users/swortain/.platformio/packages/framework-espidf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
    -- Configuring incomplete, errors occurred!
    See also "C:/Users/swortain/OneDrive/PlatformIO/220319_ESP32S3_keyboard_value/.pio/build/ESP32-S3/CMakeFiles/CMakeOutput.log".

    fatal: not a git repository (or any of the parent directories): .git
    warning: ARDUINO_SELECTIVE_WiFi (defined at C:/Users/swortain/.platformio/packages/framework-arduinoespressif32/Kconfig.projbuild:388) has direct dependencies ARDUINO_SELECTIVE_COMPILATION with value n, but is currently being y-selected by the following symbols:

    - AUTOCONNECT_WIFI (defined at C:/Users/swortain/.platformio/packages/framework-arduinoespressif32/Kconfig.projbuild:246), with value y, direct dependencies AUTOSTART_ARDUINO (value: y), and select condition AUTOSTART_ARDUINO (value: y)

    fatal: not a git repository (or any of the parent directories): .git
    fatal: not a git repository: C:/Users/swortain/.platformio/packages/framework-espidf/components/openthread/openthread/../../../.git/modules/components/openthread/openthread
    CMake Error at C:/Users/swortain/.platformio/packages/framework-arduinoespressif32/CMakeLists.txt:217 (message):
      esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)

valeros, 2023-02-14:
Hi @swortain ! The error is pretty much descriptive:
```
 esp32-arduino requires CONFIG_FREERTOS_HZ=1000 (currently 100)
```

Have you tried updating this value?

swortain, 2023-02-14:
Hi, Thanks for the reply.
emm，no....but In my opinion, maybe It should set automaticly for a new project? I delete all files in user/.platformio. then open the ide, pio download all the depends and the error appear...

Jason2866, 2023-02-14:
@swortain Would need config too. Adding a sdkconfig.defaults file is easy.
https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-arduino-blink/sdkconfig.defaults

swortain, 2023-02-17:
ok，thx a lot.
But maybe we can make a no-error sdkconfig with a specific platformio.ini automaticly as before?

valeros, 2023-02-17:
> But maybe we can make a no-error sdkconfig with a specific platformio.ini automaticly as before?

It's always been this way as default configurations are part of the ESP-IDF workflow.

closed 2023-02-17 by valeros