platformio/platform-espressif32 #1031: Can't compile Arduino as an ESP-IDF component when using PIO release v5.3.0

GKFernandes, 2023-02-14:
Hello everyone,

So, I've been trying to make a blank code example work using Arduino as an ESP-IDF Component inside Platformio.

I've followed the guidelines at: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html 

I intend to build this project to work with ESP32-C3, I'll leave a picture of my platformio.ini file, as well as the current error message when trying to build.

[image]
[image]
[image]

Many thanks!

Jason2866, 2023-02-14:
Your setup is wrong
```
framework = arduino, espidf
```
You can look at the examples how it should look like https://github.com/platformio/platform-espressif32/tree/develop/examples

GKFernandes, 2023-02-14:
Hello there @Jason2866 ,

Thank you for the response.

I've changed my setup, now it seems some Arduino function references are not being 'pulled' by the framework, Arduino setup() and loop() functions seem to not be found.

By the way, this same error was happening when using only  framework = espidf

[image]

Jason2866, 2023-02-15:
> In my project, main.cpp includes the same files as the Blink.cpp in the examples, but without the rest of the code, only the includes.

It is not the way it works. Look at the examples.

GKFernandes, 2023-02-15:
> > In my project, main.cpp includes the same files as the Blink.cpp in the examples, but without the rest of the code, only the includes.
> 
> It is not the way it works. Look at the examples.

Sorry, which part doesn't work? The file is almost exactly the same as the blink example.

I've tried build the espidf-arduino-blink example using framework = arduino, espidf, it compiles just fine, but I see it does not add the components/arduino folder, like the ESP-IDF documentation mentions.

In a previous project, I managed to use arduino as a component by creating a component folder inside the project directory and placing the Arduino libraries inside the include folder, I was using release espressif32@3.2.1 at the time and framework = espidf.

Guess I will try to use some IDF functions/properties and menuconfig to see if it works.

valeros, 2023-02-17:
> I've tried build the espidf-arduino-blink example using framework = arduino, espidf, it compiles just fine, but I see it does not add the components/arduino folder, like the ESP-IDF documentation mentions.

The Arduino component is added automatically when you specify `framework = arduino, espidf`.

> In a previous project, I managed to use arduino as a component by creating a component folder inside the project directory and placing the Arduino libraries inside the include folder, I was using release espressif32@3.2.1 at the time and framework = espidf.

Again, PlatformIO handles downloading and importing Arduino core as an IDF component when you specify `framework = arduino, espidf` in your `platformio.ini`. Your previous approach is completely possible, but in that case you're on your own as we cannot help debug your custom setup.

The recommendation here is to build your project on the basis of available examples.

closed 2023-02-17 by valeros

DrakoPOD, 2024-10-27:
I have a problem when add arduino to framework, I use an idf components "REQUIRES esp_driver_i2c" results in an error

````
fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/<userName>/.platformio/packages/framework-espidf@3.40407.240606/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'esp_driver_i2c'.
````

My CMakeList is:

```
FILE(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/src/*.*)

idf_component_register(
  SRCS ${app_sources}
  REQUIRES esp_driver_i2c
)
```

Jason2866, 2024-10-28:
Arduino needs the IDF component `esp_driver_i2c` which is not found. I don't see this driver in IDF 4.4.x so the error is as expected